balena-io/etcher #1771: Problems installing the development dependencies

shroudedcode, 2017-10-14:
As I already mentioned in my pull request #1768, I have problems getting the Etcher development "environment" to work on my machine although I closely followed the [contributing guide](https://github.com/resin-io/etcher/blob/master/docs/CONTRIBUTING.md).

### My system
- elementary OS 0.4.1 Loki (64-bit)
- Built on Ubuntu 16.04.3 LTS

### Common prerequisites
- `node --version` – `v8.7.0`
- `python --version` – `Python 2.7.12`
- `jq --version` – `jq-1.5-1-a5b5cbe`
- `curl --version` – `curl 7.47.0 [...]`

Installing the `requirements.txt` using `pip` also worked.

### Installing npm dependencies
This is where it started to get confusing. First of all the guide says that you should install the prerequisites before running the `install` script, a script which **isn't mentioned later on**. I just skipped that and ran `make electron-develop`, this threw a bunch of errors, so I [recorded it with asciinema](https://asciinema.org/a/sLIvvJV1B3OGNMA5bfjAt8yon).

After that, I tried running `npm start` anyways, but [it didn't work](https://asciinema.org/a/FTkLWQSa5zbj7bgMbbiAPM1SQ).

jhermsmeier, 2017-10-14:
Looks like libusb from our `usb` dependency blew up during installation;

<details>
<summary>
../libusb/libusb/os/linux_udev.c:40:21: fatal error: libudev.h: No such file or directory                              
compilation terminated.
</summary>

```
> usb@1.3.0 install node_modules/usb                                                                                   
> node-pre-gyp install --fallback-to-build                                                                             
                                                                                                                       
make[1]: Entering directory '/mnt/data/GitHub/etcher/node_modules/usb/build'                                           
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o                                                            
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o                                                      
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o                                                         
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o                                                              
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o                                                        
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o                                                            
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o                                                   
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o                                                
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o                                                  
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o                                                   
../libusb/libusb/os/linux_udev.c:40:21: fatal error: libudev.h: No such file or directory                              
compilation terminated.                                                                                                
libusb.target.mk:133: recipe for target 'Release/obj.target/libusb/libusb/libusb/os/linux_udev.o' failed               
make[1]: *** [Release/obj.target/libusb/libusb/libusb/os/linux_udev.o] Error 1                                         
make[1]: Leaving directory '/mnt/data/GitHub/etcher/node_modules/usb/build'                                            
gyp ERR! build error                                                                                                   
gyp ERR! stack Error: `make` failed with exit code: 2                                                                  
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)        
gyp ERR! stack     at emitTwo (events.js:125:13)                                                                       
gyp ERR! stack     at ChildProcess.emit (events.js:213:7)                                                              
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)                           
gyp ERR! System Linux 4.10.0-37-generic                                                                                
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-
to-build" "--module=/mnt/data/GitHub/etcher/node_modules/usb/src/binding/usb_bindings.node" "--module_name=usb_bindings
" "--module_path=/mnt/data/GitHub/etcher/node_modules/usb/src/binding"                                                 
gyp ERR! cwd /mnt/data/GitHub/etcher/node_modules/usb                                                                  
gyp ERR! node -v v8.7.0                                                                                                
gyp ERR! node-gyp -v v3.6.2                                                                                            
gyp ERR! not ok                                                                                                        
node-pre-gyp ERR! build error                                                                                          
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/node /usr/lib/node_modules/npm/node_modules/node-gyp/bin/nod
e-gyp.js build --fallback-to-build --module=/mnt/data/GitHub/etcher/node_modules/usb/src/binding/usb_bindings.node --mo
dule_name=usb_bindings --module_path=/mnt/data/GitHub/etcher/node_modules/usb/src/binding' (1)                         
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/node-pre-gyp/lib/util/compile.js:83:29) 
node-pre-gyp ERR! stack     at emitTwo (events.js:125:13)                                                              
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:213:7)                                                     
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:927:16)                                           
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)                   
node-pre-gyp ERR! System Linux 4.10.0-37-generic                                                                       
[... cut ...]
npm ERR!                                                                                                               
npm ERR! Failed at the usb@1.3.0 install script.                                                                       
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.                     
Makefile:511: recipe for target 'electron-develop' failed                                                              
make: *** [electron-develop] Error 1                                  
```

</details>

@jviotti any idea what might have gone wrong here?

jviotti, 2017-10-14:
Did you install libudev-dev, a build time dependency of libusb? We should probably add that to the docs.

shroudedcode, 2017-10-14:
@jviotti Seems like I haven't. After installing `libudev-dev` through `apt`, it works. That and the confusing reference to an `install` script should be fixed in the documentation.

commit ea686ee8 (2017-10-18): doc(CONTRIBUTING): Add libudev requirement to docs (#1787)

Closes #1771
Change-Type: patch
closed 2017-10-18 by jhermsmeier