balena-io/etcher #1772: Mixpanel tracking fires with disabled analytics

jhermsmeier, 2017-10-14:
- **Etcher version:** v1.1.2
- **Operating system and architecture:** any
- **Image flashed:** any
- **Do you see any meaningful error information in the DevTools?**

[image]

As can be seen here, two events are tracked and sent to Mixpanel at application startup, even though the analytics setting is disabled.

Connects To: #1770

ValdikSS, 2017-11-25:
Not fixed in 1.2.0.

jviotti, 2017-11-26:
Hi @ValdikSS ,

Thanks for reviving this thread. I thought we sent a fix for this ticket already, and that we simply forgot to close the issue with the corresponding PR. @jhermsmeier do you remember where I can find that PR?

@ValdikSS I gave this another go using the v1.2.0 on my Mac but couldn't reproduce the issue anymore. I disabled analytics, opened Etcher, and no request was sent to `api.mixpanel.com`:

[image]

How are you analyzing this? Could it be that the DevTools Network monitor is not fast enough to log certain requests you found somewhere else? Let me know so we can track this down!

jviotti, 2017-11-26:
Thanks for the clarification @ValdikSS. I was able to get to the same conclusion by using WireShark. I sent an initial PR (https://github.com/resin-io/etcher/pull/1879/files) that ensures that local settings are loaded before bootstrapping the application, but looks like the Mixpanel library still sends some initial events during its own startup that we have to find a way to shutdown. This will probably need to be done in Corvus.

jviotti, 2017-11-26:
I fixed one last mysterious `/decide/` call from the Mixpanel library here: https://github.com/resin-io-modules/resin-corvus/pull/35

jviotti, 2017-11-26:
I fixed the last remaining edge case: https://github.com/resin-io-modules/resin-corvus/pull/36.

Thanks for the heads up @ValdikSS! I definitely wouldn't have noticed that this was still the case without a packet sniffer or by refreshing the application through DevTools. 

This fix will land in the next release (probably v1.2.1) very soon, and apologies for not allocating more time to properly fix this!

ValdikSS, 2017-11-26:
Great, thanks.

closed 2017-12-05 by jviotti