balena-io/etcher #1773: Display a modal on startup when the user first runs the application, allowing the user to opt out from error reporting analytics

jviotti, 2017-10-15:
From https://github.com/resin-io/etcher/issues/1770

TheAssassin, 2017-10-15:
As mentioned in #1770, I could imagine a solution based on this idea that'd even suffice the requirements of opt-in, being easier to implement than pure opt-out.

I'd show a modal dialog like so:
[image]
(Sorry for the bad quality, I quickly hacked it together in GIMP.)

That implements the other suggestion as well, splitting up error reporting and usage tracking.

alexandrosm, 2017-10-16:
This is not something we're willing to consider doing. Etcher is built on
simplicity and usability. This goes completely against that, and would be
the first thing people would see.

closed 2017-10-16 by alexandrosm

jviotti, 2017-10-16:
Yeah, I agree that the modal is super scary. Alex's suggestion in the original thread should work, and maybe also we can provide a small tooltip right to the "Enable analytics setting" that redirects the user to a markdown file on GitHub that explains what's going on? (tied to https://github.com/resin-io/etcher/issues/1774)

strypey, 2018-04-27:
The first thing I see when I open Etcher is a modal asking me if I want to add an Etcher desktop icon, which doesn't work on Trisquel 7 (Ubuntu 14.04 base), and pops up *every time* I open the app. A modal that only pops-up the first time I open Etcher, explains why you want a program running on my PC to send you data, and gives me a choice to opt-in or not, would be an infinitely preferable UX.

Shou, 2018-04-27:
@strypey this sounds like a different issue, do you mind opening a new issue?

lurch, 2018-04-27:
> The first thing I see when I open Etcher is a modal asking me if I want to add an Etcher desktop icon

That's part of the AppImage stuff (and IIRC that modal can be disabled with an env-var). Since you're using an Ubuntu-based distro, a simple workaround would be to install Etcher from the Debian repository: https://github.com/resin-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64

lurch, 2018-04-27:
...but going back to the original topic of this PR, I recently installed VS Code (which also has usage and crash reporting enabled by default), and I thought the way they explain it on their download page was quite nice: https://code.visualstudio.com/#home-terms

strypey, 2018-04-27:
The reason given by @alexandrosm for not adding the opt-in modal was given as:
"Etcher is built on simplicity and usability. This goes completely against that, and would be the first thing people would see."

My point is that there is already a modal that doesn't serve the UX. AFAICT tweaking your build architecture to remove that, and replacing it with a thoughtfully worded and beautifully designed opt-in modal (not like the quick hack @TheAssassin threw together in GIMP ;), would improve the UX of Etcher, not damage it.

BTW VS is a proprietary *Microsoft* product. Of course it's datafarming its users. That's not remotely comparable to the situation of a free code app like Etcher.

lurch, 2018-04-27:
> BTW VS is a proprietary Microsoft product. Of course it's datafarming its users. That's not remotely comparable to the situation of a free code app like Etcher.

That's true for Visual Studio itself, but I was referring to VS Code which is an open-source project https://github.com/Microsoft/vscode
(and like Etcher, it's built on top of Electron and works on Windows, Mac and Linux)

strypey, 2018-04-27:
> I was referring to VS Code which is an open-source project 

My mistake, but remove the words "proprietary" and "free code" from that comment, and my point still stands. I'm still waiting for someone to address the UX point I'm making, which is actually relevant to this Issue.

lurch, 2018-04-27:
> I'm still waiting for someone to address the UX point I'm making

https://github.com/resin-io/etcher/issues/1773#issuecomment-384943248 and https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#deactivate-desktop-shortcut-prompt-on-gnulinux
?