balena-io/etcher #1785: MacOS's hfs images support

Corsten56, 2017-10-17:
- Etcher version: 1.1.2 x64 portable
- Operating system and architecture: Windows 8.1 x64
- Image flashed: OSX_10.13_Installer.hfs

Hello.

It is possible for Etcher to able writing HFS macOS install images?
Images downloads: https://mega.nz/#F!X0o1kCDC!Xd7LVMD8xW_yod6Z_qiAfg

or torrent magnet link:
magnet:?xt=urn:btih:31E12B5A4C94BEB607CDC42FE166F427BB160CC0&dn=CloverBootDisk&tr=http://alpha.torrenttracker.nl:443/announce

Best regards.

jhermsmeier, 2017-10-17:
@Corsten56 just a heads up here – you are aware that sharing proprietary OS images / installers in public can get you in trouble?

I'm not sure I understand what you're asking here: If Etcher already has support for this format, or if you want to suggest support for that format?

Apart from that I'm only aware of the installESD.dmg setups (see https://github.com/resin-io/etcher/issues/1572#issuecomment-313395648 for details) – which can be flashed with Etcher.

Corsten56, 2017-10-17:
Oh, sorry. Removed links. Don't think that install images restricted.

When I try to flash HFS image, Ethcher says "is not supported image type".

jhermsmeier, 2017-10-17:
If you want to try; You should be able to rename them to `filename.hfs.img` and then flash them, but as I assume that the HFS images are just file system snapshots, it'll likely not do what you want, as they're probably lacking a partition table and hence will probably not be bootable.

There are no plans to support arbitrary file system snapshots and partitioning in Etcher at the moment, so I'm going to close this for now.

closed 2017-10-17 by jhermsmeier