balena-io/etcher #1786: GPG keyserver receive failed: No data

karl69942, 2017-10-17:
- **Etcher version:** none yet
- **Operating system and architecture:** Debian 9.2 i386
- **Image flashed:** none yet
- **Do you see any meaningful error information in the DevTools?** nope...??

 sudo apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
Executing: /tmp/apt-key-gpghome.4JhYetPTcO/gpg.1.sh --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
gpg: keyserver receive failed: No data

W: GPG error: https://dl.bintray.com/resin-io/debian stable Release: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 379CE192D401AB61
E: The repository 'https://dl.bintray.com/resin-io/debian stable Release' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

I've been having some unrelated driver issues on my netbook so I've had to re-install Debian a few times, when I tried to add the etcher repo this is what I ran into above ^  
Any ideas how I can get this thing working?

karl69942, 2017-10-17:
Oh that's funny, its working now. not sure why it kept failing. Sorry! Can't understand why it would just suddenly start to work again...  ah well!

lurch, 2017-10-17:
Might just have been a temporary failure on the pgp.mit.edu server?

closed 2017-10-17 by lurch

karl69942, 2017-10-17:
Honestly I am not sure. I kept trying it & it kept failing, I came back to it a little later, just after making this thread, and it suddenly started working.. Weird.

4joey1, 2018-04-07:
Mine still fails on Debian 9 stretch 64 bit as below.

W: GPG error: https://dl.bintray.com/resin-io/debian stable Release: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 379CE192D401AB61
W: The repository 'https://dl.bintray.com/resin-io/debian stable Release' is not signed.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.

Kindly assist.

4joey1, 2018-04-07:
I ran the command below and no errors appear after running the apt-get update command

sudo apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61

djkerya, 2019-01-10:
sudo apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
Executing: /tmp/apt-key-gpghome.3wzLSL1SLd/gpg.1.sh --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
gpg: keyserver receive failed: No data

maximeaudy, 2019-01-10:
I have the same error @djkerya

lurch, 2019-01-10:
@djkerya @maximeaudy Looks like you're following outdated instructions? https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64 uses the Ubuntu keyserver rather than the MIT keyserver.

lurch, 2019-01-12:
@djkerya Yes, I did see your comment. You seem to be trying to use `hkp://pgp.mit.edu:80` but the instructions at https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64 suggest using `keyserver.ubuntu.com` (the instructions were changed by #2319 in May 2018).