balena-io/etcher #1788: USB flash drive not detected on macOS

shpokas, 2017-10-18:
- **Etcher version:**
v1.1.2
- **Operating system and architecture:**
macOS 10.13 (17A405)
- **Image flashed:**
pfSense-CE-memstick-serial-2.4.0-RELEASE-amd64.img from https://www.pfsense.org/download/
- **Do you see any meaningful error information in the DevTools?**
No.

This issue looks vaguely similar to #649, at least same USB manufacturer and same OS vendor.
**diskutil output**

imbp:draza $ diskutil info /dev/disk2
   Device Identifier:        disk2
   Device Node:              /dev/disk2
   Whole:                    Yes
   Part of Whole:            disk2
   Device / Media Name:      Extreme

   Volume Name:              Not applicable (no file system)
   Mounted:                  Not applicable (no file system)
   File System:              None

   Content (IOContent):      FDisk_partition_scheme
   OS Can Be Installed:      No
   Media Type:               Generic
   Protocol:                 USB
   SMART Status:             Not Supported

   Disk Size:                15.7 GB (15693664256 Bytes) (exactly 30651688 512-Byte-Units)
   Device Block Size:        512 Bytes

   Read-Only Media:          No
   Read-Only Volume:         Not applicable (no file system)

   Device Location:          External
   **Removable Media:          Fixed**

   Virtual:                  No
   OS 9 Drivers:             No
   Low Level Format:         Not supported

**Full darwin.sh output:**
imbp:draza $ bash ./darwin.sh 
device: /dev/disk0
displayName: /dev/disk0
description: "APPLE SSD SM0512G"
size: 500277790720
mountpoints: []
raw: /dev/rdisk0
protected: False
system: True

device: /dev/disk1
displayName: /dev/disk1
description: "APPLE SSD SM0512G"
size: 499418034176
mountpoints:
  - path: "/"
  - path: "/private/var/vm"
raw: /dev/rdisk1
protected: False
system: True

device: /dev/disk2
displayName: /dev/disk2
description: "Extreme"
size: 15693664256
mountpoints:
  - path: "/Volumes/HPE-ESXI-5_"
raw: /dev/rdisk2
protected: False
system: True

shpokas, 2017-10-18:
GUI Diskutil gives this information for USB drive.

Volume type : Physical Device
BSD device node : disk2
Connection : USB
Device tree path : IODeviceTree:/PCI0@0/XHC1@14
Writable : No
Is case-sensitive : No
Volume capacity : 15 693 664 256
Available space (Purgeable + Free) : 0
Purgeable space : 0
Free space : 0
Used space : 15 693 664 256
Owners enabled : No
Is encrypted : No
Can be verified : No
Can be repaired : No
Bootable : No
Journaled : No
Disk number : 2
Media name : SanDisk Extreme Media
Media type : Generic
Ejectable : Yes
Solid state : No
S.M.A.R.T. status : Not Supported

lurch, 2017-10-18:
If Apple's `diskutil` is telling us `Removable Media: Fixed` then I think the only safe thing for `drivelist` to do is report it as a "system" drive. And in Etcher you can write to system drives after enabling "Unsafe mode".
(apologies if you knew all that already)

closed 2017-12-04 by jviotti