balena-io/etcher #1792: invaliod block type error

hbmau, 2017-10-18:
- **Etcher version:**etcher 1.1.2
- **Operating system and architecture:**Windows 10 64 Bit
- **Image flashed:**rasbian stretch
- **Do you see any meaningful error information in the DevTools?**No  on first tempt  i could open dev tool with Ctr +Alt+ I, later DEv Tools did not open again

Hi, I used several micro SD cards (16GB, 32 GB: always the same error: invalid block type at Zlib._handle.onerror (zlib.js:355:17) , the same error at the different micros SD cards. Cleaned the cards with diskpart, the same error occured: how solve this problem? Other Programm? (DiskImager?)
Thanks for answering
my EMAIL: praxismaurer@bluewin.ch

lurch, 2017-10-18:
Can you please verify the _exact_ version of Raspbian Stretch that you're using (a download link is ideal) as that'll make it easier for us to investigate this problem.

hbmau, 2017-10-19:
Thanks  for your answer.

I used the newest version from www..raspberrypi.org/downloads/Raspbian: Version September 2017,release 2017-09-07,Kernel Version 4.9

 

Sincerely yours

 

H.Maurer

jhermsmeier, 2017-10-19:
@hbmau thanks, we'll have a look!

hbmau, 2017-10-19:
I’d like to inform you about the following:

I just tried etcher with an external USB Card writer, the same problem, with the same error announcement. So it is not the older build in card reader of my PC.

Mabee some (Micro) SD Cards don’t function orderly?

 

I’l be really excited about a possible solution of this problem!

Sincerely yours   

H.Maurer

jhermsmeier, 2017-10-19:
This has nothing to do with the SD-cards or their adapters – the error is about the compressed blocks in the .zip file; I've now tested both, Raspbian Stretch w/ Desktop and Stretch Lite straight from the .zip, and both worked just fine.

My guess here is that your downloads have corrupted due to network issues or so – you should be able to check your downloaded images against the checksums provided on https://www.raspberrypi.org/downloads/raspbian/.

lurch, 2017-10-19:
And if you're unsure how to varify checksums, I just did a quick search and found https://bhoover.com/how-to-verify-checksum-windows/

closed 2017-12-04 by jviotti