balena-io/etcher #1793: Couldn't clean the drive, Command Failed: diskpart /s   --- WIn 8.1

MountainLogic, 2017-10-19:
- **Etcher version:** 1.1.2
- **Operating system and architecture:** WIn 8.1 on bootcamp using no-name external reader 
- **Image flashed:** Beagle bone black Ubuntu image
- **Note,** in Windows Explorer, the drive icon & lebel "boot (e:)" on left appeared and disappeared several times before error message.  Also Viper anti-virus scanned with each cycle.  
- **Note,** Attempted several times, including as admin with same results.  Also tried with anti-virus turned off with no success.  Also tried in unsafe mode with no success

- **Do you see any meaningful error information in the DevTools?**  
Couldn't clean the drive, Command Failed: diskpart

[image]

jhermsmeier, 2017-10-19:
ref #1054

jhermsmeier, 2017-10-19:
Thanks for the additional info on the issue – we thought we'd squashed it already

lurch, 2017-10-19:
@jhermsmeier It looks like #989 is a better duplicate for this than #1054 ;-)
#1054 is about the `.bat` file failing, whereas this PR and #989 are specifically about `diskpart /s` failing. But I guess this is another one of those problems that'll be fixed when we manage to get native drive-locking into place?

jhermsmeier, 2017-10-19:
Hmm, yeah, maybe – this doesn't fail due to a timeout or running out of retries – might be entirely new thing. But yes, this could possibly be fixed by drive-locking.

cybertonian, 2017-10-24:
I am getting the "Couldn't clean the drive..." (code: 2147942401) on two different drives.  The only difference between when this used to work and now is that I installed the latest patches to Windows10 (KB4043901).  Is there an issue with this MS update?

MountainLogic, 2017-10-24:
FWIW, I started to notice that Windows kept asking with a pop-up to format the drive and just before I saw the pop-up, is when I got the error in etcher.  Also saw an generic write error when using Win32DiskImager and Rufus.  Eventually, I kept thrashing and was able to get it flashed with Win32DiskImager.  Not sure how.

DuaneEllis-TI, 2017-11-21:
FYI - I had the exact same problem

In my case - this is what happened and how I fixed it.

In my case I am using  FULL SIZED sd card adapter
With a small micro sd card installed in the adapter.

The etcher tool is creating a TEMP file with commands for "diskpart"
then executing the "diskpart" tool telling it to read from the file
If you run the same command and manually the the commands
you discover that the DISK WRITE protect tab is done

What Etcher needs is some means *read* the WRITE PROTECT status of the SD CARD
And stop/halt or give an error before proceeding.
Or maybe it has a means but that no longer works.

Moving the Write Protect switch solves the problem.

Dumbleboar, 2017-11-24:
I also solved this error by using a different Adapter Card. 
Keep in mind that some sd card adapters have a small switch that locks the card from being written to

lurch, 2017-11-28:
> What Etcher needs is some means read the WRITE PROTECT status of the SD CARD

Etcher should display a "LOCKED" warning for read-only SD cards https://github.com/resin-io/etcher/pull/475

tjeulink, 2017-12-15:
Had the same error, the TF card to SD card adapter had locked without me noticing. It didn't display locked or anything like it. 

Was flashing latest raspbian on 32gb samsung evo TFsd card with the included adapter on a non administrator account. had to give administrative privileges in compatibility tab for it to display this message otherwise it would've been stuck on starting without giving any clue whats going on.

HakanKoseoglu, 2017-12-21:
Looks like Windows is accessing the existing file systems, at least in my case. Removing all partitions from the SD card and retrying works fine after this error.

cymplecy, 2018-01-09:
Same error - caused by SD Card adaptor lock switch - all OK after unlocking

lurch, 2018-01-09:
@cymplecy What SD Card reader and OS are you using? I wonder if it's possible that `drivelist` (used by Etcher to list drives) isn't detecting that the SD card is locked, even though the OS _does_ still prevent writing to it?

cymplecy, 2018-01-09:
This on win10 with the adaptor that came with the micro sd

or mute
the thread

DuaneEllis-TI, 2018-01-09:
In my case it is  a 'micro-usb' to 'standard-size-adapter' 

the standardsized interface is built into my dell E7450 laptop  on the side there is a slot to insert a standard sized sd card

That interface is not an USB controller, if you follow the device manager "view by connection" it connects via the PCI interface.

lurch, 2018-01-10:
Thanks for the info @DuaneEllis-TI and @cymplecy I'll do some investigating to see if I can reproduce.

DuaneEllis-TI, 2018-01-10:
FYI - the following works just fine on my windows machine
I get this output:

  * Case 1 --> Success: 1, Flags: 0x00080006  WP=Yes
  * Case 2 --> Success: 1, Flags: 0x00000006  WP=No
```
#include "stdafx.h"
#include "windows.h"

int main()
{
	char drive[] = "d:\\";
	BOOL success;
	char vol_name[1024];
	DWORD serialnumber;
	DWORD maxlength;
	DWORD flags;
	char sysnamebuf[1024];

	success = GetVolumeInformationA(
		drive,
		vol_name,
		sizeof(vol_name),
		&serialnumber,
		&maxlength,
		&flags,
		sysnamebuf,
		sizeof(sysnamebuf));
	char *wp = (flags & FILE_READ_ONLY_VOLUME) ? "Yes" : "No";
	printf("Success: %d, Flags: 0x%08x  WP=%s\n", success, flags, wp);
		
    return 0;
}

```

allengotstuff, 2018-01-13:
@lurch  I just brought my first raspberry pi 3 today, and i had the same problem when i tried to flash the Raspbian OS on to my newly purchased min SD card with a adapter that has a lock switch. I just found out that even if the locked switch is off, the computer is still think the SD card is locked. therefore, i saw this error. But when i purchased a new mini SD card adapter, and everything was ok.

So sometime the SD card adapter hardware malfunctions, check if the lock is really off but try to copy sample file to it.

lurch, 2018-02-28:
Could the people who've commented in this issue please test this version of Etcher to see if it fixes your problems? Thanks :)
https://resin-nightly-downloads.s3.amazonaws.com/etcher/2018-02-20/1.3.1%2B5be64e1/Etcher-Portable-1.3.1%2B5be64e1-x64.exe

cymplecy, 2018-02-28:
Tried it - enabled lock switch - got an error msgs that starts off "operation timed out...."

cymplecy, 2018-02-28:
Oh dear - doesn't work even though I've taken lock switch off!  Going to try old version just to make sure its a software issue

cymplecy, 2018-02-28:
Old version 1.3.1-x64 working fine

jhermsmeier, 2018-02-28:
@cymplecy I've now tested this build on both Windows 7 and Windows 10 (unfortunately I do not have Windows 8.1 at my disposal atm), and cannot reproduce this – can you post screenshots of those errors? And can you open the DevTools (Ctrl+Shift+I), and post what the console shows?

cymplecy, 2018-02-28:
[image]

ctrl+shift+l doesn nothing (is it one or i or little L?)

cymplecy, 2018-02-28:
I'm on Win10 - this is with USB adaptor - with SD Adaptor - with 32GB microSD inside that

jhermsmeier, 2018-02-28:
Oh sorry, it's a capital i; <kbd>Ctrl</kbd>+<kbd>Shift</kbd>+<kbd>I</kbd>

jhermsmeier, 2018-02-28:
Thanks for the screenshot, I now have a rough idea of what might be going on there – still some console logs from the DevTools could help

cymplecy, 2018-02-28:
I had to close the error before the console came up :)

`C:\Users\Simon\AppData\Local\Temp\nsh7692.tmp\app\resources\app.asar\node_modules\debug\src\browser…:133 sdk:scanner scanner:init +0ms Scanner
C:\Users\Simon\AppData\Local\Temp\nsh7692.tmp\app\resources\app.asar\node_modules\debug\src\browser…:133 sdk:scanner subscribe +1ms BlockDeviceAdapter
C:\Users\Simon\AppData\Local\Temp\nsh7692.tmp\app\resources\app.asar\node_modules\debug\src\browser…:133 sdk:scanner subscribe +1ms USBBootAdapter
gui.js:65102  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@resin.io

Version = 1.3.1, Type = local
gui.js:113926 Wed Feb 28 2018 17:40:24 GMT+0000 (GMT Standard Time) Application start ({"packageType":"local","version":"1.3.1"})
gui.js:113926 Wed Feb 28 2018 17:40:24 GMT+0000 (GMT Standard Time) Checking for updates ({"currentVersion":"1.3.1","stable":true,"updateSemverRange":"<2.0.0","includeUnstableChannel":false})
C:\Users\Simon\AppData\Local\Temp\nsh7692.tmp\app\resources\app.asar\node_modules\debug\src\browser…:133 sdk:scanner start +125ms true
gui.js:113926 Wed Feb 28 2018 17:40:25 GMT+0000 (GMT Standard Time) [object Event]
gui.js:113926 Wed Feb 28 2018 17:40:26 GMT+0000 (GMT Standard Time) Update notification skipped ({"reason":"Latest version"})
gui.js:113926 Wed Feb 28 2018 17:40:26 GMT+0000 (GMT Standard Time) Open image selector
gui.js:113926 Wed Feb 28 2018 17:40:33 GMT+0000 (GMT Standard Time) Select image ({"path":"C:\\Users\\Simon\\Downloads\\2017-11-29-raspbian-stretch.zip","size":{"original":4919918592,"final":{"estimation":false,"value":4919918592}},"extension":"img","archiveExtension":"zip","hasMBR":true,"hasGPT":false,"partitions":[{"type":12,"id":null,"name":null,"firstLBA":8192,"lastLBA":93237,"extended":false},{"type":131,"id":null,"name":null,"firstLBA":94208,"lastLBA":9609216,"extended":false},{"type":0,"id":null,"name":null,"firstLBA":0,"lastLBA":0,"extended":false},{"type":0,"id":null,"name":null,"firstLBA":0,"lastLBA":0,"extended":false}],"logo":false,"bmap":false})
C:\Users\Simon\AppData\Local\Temp\nsh7692.tmp\app\resources\app.asar\node_modules\debug\src\browser…:133 sdk:scanner stop +12s true
gui.js:113926 Wed Feb 28 2018 17:40:36 GMT+0000 (GMT Standard Time) Progress (undefined): 0% at 0 MB/s (eta undefineds)
gui.js:113926 Wed Feb 28 2018 17:40:36 GMT+0000 (GMT Standard Time) Flash ({"image":"C:\\Users\\Simon\\Downloads\\2017-11-29-raspbian-stretch.zip","drive":{"size":32026656768,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"H:\\","blockSize":512,"adapter":"blockdevice","isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive1","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"mountpoints":[{"path":"H:\\"}],"description":"Mass Storage Device USB Device","isUAS":false},"uuid":"3c7e8c92-2b56-437f-b7fc-9f1d00fdaadb","unmountOnSuccess":true,"validateWriteOnSuccess":true})
gui.js:146348 Elevating command: C:\Users\Simon\AppData\Local\Temp\nsh7692.tmp\app\Etcher.exe C:\Users\Simon\AppData\Local\Temp\nsh7692.tmp\app\resources\app.asar
gui.js:146315 Successfully connected to IPC server: etcher-server-8756, socket root C:\Users\Simon\AppData\Local\Temp\
gui.js:146315 Image: C:\Users\Simon\Downloads\2017-11-29-raspbian-stretch.zip
gui.js:146315 Device: \\.\PhysicalDrive1
gui.js:146315 Umount on success: true
gui.js:146315 Validate on success: true
gui.js:113926 Wed Feb 28 2018 17:40:42 GMT+0000 (GMT Standard Time) Flash error ({"error":{"message":"operation timed out after 4006 ms, 5 tries with error: EPIPE, Couldn't open handle to device '\\\\.\\PhysicalDrive1'","stack":"Error: operation timed out after 4006 ms, 5 tries with error: EPIPE, Couldn't open handle to device '\\\\.\\PhysicalDrive1'\n    at n    at tryCatcher (C:\\Users\\Simon\\AppData\\Local\\Temp\\nsh7692.tmp\\app\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\util.js:16:23)\n    at n    at tryCatcher (C:\\Users\\Simon\\AppData\\Local\\Temp\\nsh7692.tmp\\app\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\util.js:16:23)\n    at Promise._settlePromiseFromHandler (C:\\Users\\Simon\\AppData\\Local\\Temp\\nsh7692.tmp\\app\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\promise.js:504:31)\n    at Promise._settlePromise (C:\\Users\\Simon\\AppData\\Local\\Temp\\nsh7692.tmp\\app\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\promise.js:561:18)\n    at Promise._settlePromise0 (C:\\Users\\Simon\\AppData\\Local\\Temp\\nsh7692.tmp\\app\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\promise.js:606:10)\n    at Promise._settlePromises (C:\\Users\\Simon\\AppData\\Local\\Temp\\nsh7692.tmp\\app\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\promise.js:681:18)\n    at Async._drainQueue (C:\\Users\\Simon\\AppData\\Local\\Temp\\nsh7692.tmp\\app\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\async.js:138:16)\n    at Async._drainQueues (C:\\Users\\Simon\\AppData\\Local\\Temp\\nsh7692.tmp\\app\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\async.js:148:10)\n    at Immediate.Async.drainQueues (C:\\Users\\Simon\\AppData\\Local\\Temp\\nsh7692.tmp\\app\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\async.js:17:14)\n    at runCallback (tim [... cut ...] 672:20)\n    at tryOnImmediate (timers.js:645:5)\n    at processImmediate [as _immediateCallback] (timers.js:617:5)","message":"operation timed out after 4006 ms, 5 tries with error: EPIPE, Couldn't open handle to device '\\\\.\\PhysicalDrive1'","code":"ETIMEDOUT"}
logException @ gui.js:114115
C:\Users\Simon\AppData\Local\Temp\nsh7692.tmp\app\resources\app.asar\node_modules\debug\src\browser…:133 sdk:scanner start +26m true
gui.js:146373 Flash results Object
gui.js:146394 Terminating IPC server`

jhermsmeier, 2018-02-28:
This is great, thanks! Looks like we have a lead on the rather elusive ETIMEDOUT bug here :)

PS: Edited your comment to display the preformatted output correctly

cymplecy, 2018-02-28:
JFI - this is without the adaptor being locked - this is just trying to use it normally

yoonghm, 2018-03-02:
Below is log for the same issue:

```
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 sdk scanner:init +0ms Scanner
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 sdk subscribe +2ms StandardAdapter
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 sdk subscribe +1ms USBBootAdapter
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@resin.io

Version = 1.3.1, Type = portable
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 Fri Mar 02 2018 15:58:06 GMT+0800 (Malay Peninsula Standard Time) Application start ({"packageType":"portable","version":"1.3.1"})
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 Fri Mar 02 2018 15:58:06 GMT+0800 (Malay Peninsula Standard Time) Checking for updates ({"currentVersion":"1.3.1","stable":true,"updateSemverRange":"<2.0.0","includeUnstableChannel":false})
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 sdk start +538ms true
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 Fri Mar 02 2018 15:58:12 GMT+0800 (Malay Peninsula Standard Time) Update notification skipped ({"reason":"Latest version"})
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 Fri Mar 02 2018 15:58:20 GMT+0800 (Malay Peninsula Standard Time) Open image selector
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 Fri Mar 02 2018 15:58:22 GMT+0800 (Malay Peninsula Standard Time) Select image ({"path":"D:\\KBS\\Raspberry_Pi\\Bin\\2017-11-29-raspbian-stretch.img","extension":"img","size":{"original":4919918592,"final":{"estimation":false,"value":4919918592}},"hasMBR":true,"hasGPT":false,"partitions":[{"type":12,"id":null,"name":null,"firstLBA":8192,"lastLBA":93237,"extended":false},{"type":131,"id":null,"name":null,"firstLBA":94208,"lastLBA":9609216,"extended":false},{"type":0,"id":null,"name":null,"firstLBA":0,"lastLBA":0,"extended":false},{"type":0,"id":null,"name":null,"firstLBA":0,"lastLBA":0,"extended":false}],"logo":false,"bmap":false})
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 Fri Mar 02 2018 15:58:24 GMT+0800 (Malay Peninsula Standard Time) Open modal ({"template":"./components/drive-selector/templates/drive-selector-modal.tpl.html"})
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 Fri Mar 02 2018 15:58:24 GMT+0800 (Malay Peninsula Standard Time) Reselect drive
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 Fri Mar 02 2018 15:58:25 GMT+0800 (Malay Peninsula Standard Time) Toggle drive ({"drive":{"size":15987638272,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"E:\\, "}],"description":"SDHC Card","isUAS":false},"previouslySelected":true})
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 Fri Mar 02 2018 15:58:27 GMT+0800 (Malay Peninsula Standard Time) Modal accepted ({})
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 sdk stop +21s true
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 Fri Mar 02 2018 15:58:28 GMT+0800 (Malay Peninsula Standard Time) Progress (undefined): 0% at 0 MB/s (eta undefineds)
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 Fri Mar 02 2018 15:58:28 GMT+0800 (Malay Peninsula Standard Time) Flash ({"image":"D:\\KBS\\Raspberry_Pi\\Bin\\2017-11-29-raspbian-stretch.img","drive":{"size":15987638272,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"E:\\, "}],"description":"SDHC Card","isUAS":false},"uuid":"d83aa419-04c9-4c4b-a941-6ed54e82ad20","unmountOnSuccess":true,"validateWriteOnSuccess":true})
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 Elevating command: C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\Etcher.exe C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 Successfully connected to IPC server: etcher-server-1476, socket root C:\Users\yhm2\AppData\Local\Temp\
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 Image: D:\KBS\Raspberry_Pi\Bin\2017-11-29-raspbian-stretch.img
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 Device: \\.\PhysicalDrive1
[... cut ...]
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 sdk start +2m true
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 Flash results Object
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 Terminating IPC server
C:\Users\yhm2\AppData\Local\Temp\nsyFB27.tmp\app\resources\app.asar\node_modules\raven-js\dist\rave…:55 Fri Mar 02 2018 16:00:54 GMT+0800 (Malay Peninsula Standard Time) Modal accepted ({"value":false})
```

lurch, 2018-03-02:
> Looks like we have a lead on the rather elusive ETIMEDOUT bug

IIRC this is the first time that we've had the ETIMEDOUT error on an image that _isn't_ one of the bz2 Hassio images?

cybertonian, 2018-03-03:
Version = 1.3.1, Type = localgui.js:113926 Fri Mar 02 2018 16:27:51 GMT-0800 (Pacific Standard Time) Application start ({"packageType":"local","version":"1.3.1"})gui.js:113926 Fri Mar 02 2018 16:27:51 GMT-0800 (Pacific Standard Time) Checking for updates ({"currentVersion":"1.3.1","stable":true,"updateSemverRange":"<2.0.0","includeUnstableChannel":false})C:\Users\egk\AppData\Local\Temp\nsw5739.tmp\app\resources\app.asar\node_modules\debug\src\browser.js:133 sdk:scanner start +251ms truegui.js:113926 Fri Mar 02 2018 16:27:54 GMT-0800 (Pacific Standard Time) Update notification skipped ({"reason":"Latest version"})gui.js:113926 Fri Mar 02 2018 16:27:55 GMT-0800 (Pacific Standard Time) [object Event]gui.js:113926 Fri Mar 02 2018 16:28:00 GMT-0800 (Pacific Standard Time) Open image selectorgui.js:113926 Fri Mar 02 2018 16:45:47 GMT-0800 (Pacific Standard Time) Image selector closedgui.js:113926 Fri Mar 02 2018 16:46:03 GMT-0800 (Pacific Standard Time) Open image selectorgui.js:113926 Fri Mar 02 2018 16:46:09 GMT-0800 (Pacific Standard Time) Select image ({"path":"C:\\Users\\egk\\Downloads\\2017-11-29-raspbian-stretch-lite.zip","size":{"original":1858076672,"final":{"estimation":false,"value":1858076672}},"extension":"img","archiveExtension":"zip","hasMBR":true,"hasGPT":false,"partitions":[{"type":12,"id":null,"name":null,"firstLBA":8192,"lastLBA":93237,"extended":false},{"type":131,"id":null,"name":null,"firstLBA":94208,"lastLBA":3629056,"extended":false},{"type":0,"id":null,"name":null,"firstLBA":0,"lastLBA":0,"extended":false},{"type":0,"id":null,"name":null,"firstLBA":0,"lastLBA":0,"extended":false}],"logo":false,"bmap":false})C:\Users\egk\AppData\Local\Temp\nsw5739.tmp\app\resources\app.asar\node_modules\debug\src\browser.js:133 sdk:scanner stop +18m truegui.js:113926 Fri Mar 02 2018 16:46:13 GMT-0800 (Pacific Standard Time) Progress (undefined): 0% at 0 MB/s (eta undefineds)gui.js:113926 Fri Mar 02 2018 16:46:13 GMT-0800 (Pacific Standard Time) Flash ({"image":"C:\\Users\\egk\\Downloads\\2017-11-29-raspbian-stretch-lite.zip","drive":{"size":15833497600,"isVirtual":false,"enumerator":"SCSI","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"E:\\","blockSize":512,"adapter":"blockdevice","isSCSI":true,"isRemovable":true,"device":"\\\\.\\PhysicalDrive1","busVersion":"2.0","isSystem":false,"busType":"INVALID","isCard":false,"isUSB":false,"mountpoints":[{"path":"E:\\"}],"description":"O2Micro SD SCSI Disk Device","isUAS":false},"uuid":"d2c90b4c-f147-4d5b-b9c4-3378785d4c77","unmountOnSuccess":false,"validateWriteOnSuccess":true})gui.js:146348 Elevating command: C:\Users\egk\AppData\Local\Temp\nsw5739.tmp\app\Etcher.exe C:\Users\egk\AppData\Local\Temp\nsw5739.tmp\app\resources\app.asargui.js:146315 Successfully connected to IPC server: etcher-server-5620, socket root C:\Users\egk\AppData\Local\Temp\gui.js:146315 Image: C:\Users\egk\Downloads\2017-11-29-raspbian-stretch-lite.zipgui.js:146315 Device: \\.\PhysicalDrive1gui.js:146315 Umount on success: falsegui.js:146315 Validate on success: truegui.js:113926 Fri Mar 02 2018 16:46:23 GMT-0800 (Pacific Standard Time) Flash error ({"error":{"message":"operation timed out after 4067 ms, 5 tries with error: EPERM, Couldn't delete drive layout '\\\\.\\PhysicalDrive1'","stack":"Error: operation timed out after 4067 ms, 5 tries with error: EPERM, Couldn't delete drive layout '\\\\.\\PhysicalDrive1'\n    at n    at tryCatcher (C:\\Users\\egk\\AppData\\Local\\Temp\\nsw5739.tmp\\app\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\util.js:16:23)\n    at n    at tryCatcher (C:\\Users\\egk\\AppData\\Local\\Temp\\nsw5739.tmp\\app\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\util.js:16:23)\n    at Promise._settlePromiseFromHandler (C:\\Users\\egk\\AppData\\Local\\Temp\\nsw5739.tmp\\app\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\promise.js:504:31)\n    at Promise._settlePromise (C:\\Users\\egk\\AppData\\Local\\Temp\\nsw5739.tmp\\app\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\promise.js:561:18)\n    at Promise._settlePromise0 (C:\\Users\\egk\\AppData\\Local\\Temp\\nsw5739.tmp\\app\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\promise.js:606:10)\n    at Promise._settlePromises (C:\\Users\\egk\\AppData\\Local\\Temp\\nsw5739.tmp\\app\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\promise.js:681:18)\n    at Async._drainQueue (C:\\Users\\egk\\AppData\\Local\\Temp\\nsw5739.tmp\\app\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\async.js:138:16)\n    at Async._drainQueues (C:\\Users\\egk\\AppData\\Local\\Temp\\nsw5739.tmp\\app\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\async.js:148:10)\n    at Immediate.Async.drainQueues (C:\\Users\\egk\\AppData\\Local\\Temp\\nsw5739.tmp\\app\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\async.js:17:14)\n    at runCallback (timers.js:672:20)\n    at tryOnImmediate (timers.js:645:5)\n    at processImmediate [as _immediateCallback] (timers.js:617:5)","code":"ETIMEDOUT"},"image":"C:\\Users\\egk\\Downloads\\2017-11-29-raspbian-stretch-lite.zip","drive":{"size":15833497600,"isVirtual":false,"enumerator":"SCSI","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"E:\\","blockSize":512,"adapter":"blockdevice","isSCSI":true,"isRemovable":true,"device":"\\\\.\\PhysicalDrive1","busVersion":"2.0","isSystem":false,"busType":"INVALID","isCard":fa [... cut ...] "code":"ETIMEDOUT"}logException @ gui.js:114115C:\Users\egk\AppData\Local\Temp\nsw5739.tmp\app\resources\app.asar\node_modules\debug\src\browser.js:133 sdk:scanner start +15s truegui.js:146373 Flash results Objectgui.js:146394 Terminating IPC servergui.js:113926 Fri Mar 02 2018 16:53:44 GMT-0800 (Pacific Standard Time) Modal accepted ({"name":"warning","value":false})

   

https://resin-nightly-downloads.s3.amazonaws.com/etcher/2018-02-20/1.3.1%2B5be64e1/Etcher-Portable-1.3.1%2B5be64e1-x64.exe—

yoonghm, 2018-03-05:
Windows Defender does not allow me to run this app.  Something is wrong.  Do you have alternatives?

jhermsmeier, 2018-03-05:
@yoonghm you mean the snapshot build linked to above? That's probably because its not codesigned (we don't codesign snapshots) – you'll want to download the current stable release instead.

yoonghm, 2018-03-05:
@jhermsmeier actually I was using the latest stable version when the problem occurred. The post above asked me to try the snapshot. Is it safe to run?

jhermsmeier, 2018-03-05:
@yoonghm oh, right – well, what is Windows Defender saying? Can you post a screenshot?

yoonghm, 2018-03-06:
[image]

[image]

jhermsmeier, 2018-03-06:
Yeah, that warning is just due to lack of code signing – glad it works for you now @yoonghm :)

jhermsmeier, 2018-04-06:
Fixed in #2023

shelllter, 2018-04-13:
FWIW, I was getting this error message, but when I checked the status of the uSD card with Minitool Partition Wizard, it reported "Bad Disk".

When I used an uSD card that passed the Partition Wizard scan, Etcher worked as well :-) .

Doah!

lurch, 2018-04-20:
Should this be re-opened, given that #2023 has been reverted?

jhermsmeier, 2018-04-21:
Yeah, as it turned out it's a bit more involved than initially thought, and didn't cover enough cases of disk formats / states in order to reliable "clean" them. This should be re-closed again soon-ish though :)

Sunnysidedown916, 2018-05-26:
I originally had a USB hub on my card reader, I connected the card reading to the USB port on my PC and this fixed it.

Unixware, 2018-06-09:
epic bug still effin exist wtf

Applejackmw, 2018-07-06:
Same issue. The portable Etcher worked for mine. Started writing straight away. Thanks for the help.

jkw, 2018-07-27:
Just encountered this error. Managed to flash yesterday with Etcher and same USB-stick.

Image of error popup:
[image]

Backtrace from dev tools:
`{"stack":"Error: Couldn't clean the drive, Command failed: diskpart /s C:\\Users\\kwedm\\AppData\\Local\\Temp\\etcher-diskpart-40953b54fa04\n (code 2147942405)\n    at retry.catch (C:\\Users\\kwedm\\AppData\\Local\\Programs\\etcher\\resources\\app.asar\\lib\\cli\\diskpart.js:117:15)\n    at tryCatcher (C:\\Users\\kwedm\\AppData\\Local\\Programs\\etcher\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\util.js:16:23)\n    at Promise._settlePromiseFromHandler (C:\\Users\\kwedm\\AppData\\Local\\Programs\\etcher\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\promise.js:504:31)\n    at Promise._settlePromise (C:\\Users\\kwedm\\AppData\\Local\\Programs\\etcher\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\promise.js:561:18)\n    at Promise._settlePromise0 (C:\\Users\\kwedm\\AppData\\Local\\Programs\\etcher\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\promise.js:606:10)\n    at Promise._settlePromises (C:\\Users\\kwedm\\AppData\\Local\\Programs\\etcher\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\promise.js:681:18)\n    at Async._drainQueue (C:\\Users\\kwedm\\AppData\\Local\\Programs\\etcher\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\async.js:138:16)\n    at Async._drainQueues (C:\\Users\\kwedm\\AppData\\Local\\Programs\\etcher\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\async.js:148:10)\n    at Immediate.Async.drainQueues (C:\\Users\\kwedm\\AppData\\Local\\Programs\\etcher\\resources\\app.asar\\node_modules\\bluebird\\js\\release\\async.js:17:14)\n    at runCallback (timers.js:672:20)\n    at tryOnImmediate (timers.js:645:5)\n    at processImmediate [as _immediateCallback] (timers.js:617:5)","message":"Couldn't clean the drive, Command failed: diskpart /s C:\\Users\\kwedm\\AppData\\Local\\Temp\\etcher-diskpart-40953b54fa04\n (code 2147942405)","name":"Error"}
console.(anonymous function) @ gui.js:105081
`

nineff, 2018-08-04:
I'm getting a sdimilar error with Etcher 1.4.4 on Win7
[image]
The small german part means essentially "command 'diskpart' not found"

frankgorhamengard, 2018-09-14:
I found a solution that worked for me.
I had same problem. Same error message. SD micro card was plugged in to a USB adapter when it was formatted by SD formatter. Even when I moved it to SD card adapter for regular SD size it still failed. But after I reformatted in the SD adapter, I was able to install the image successfully with Etcher.
Something about formatting in a USB adapter?

Win 10 64.  Etcher 1.4.4   SD_CardFormatter0500

cyoung, 2019-02-08:
This is still an issue in 1.4.9.

When looking at the disk in "Computer Management", Windows states:

"Offline (The disk is offline because it has a signature collision with another disk that is online)"

After using "Computer Management" to remove the partitions, then disconnecting and re-connecting the USB-to-microUSB adapter, Etcher was able to flash the image on the microSD card.

RickBullotta, 2019-03-02:
FWIW, I think the issue is related to some type of compatibility problem with the O2Micro MMC/SD controller.

closed 2023-07-12 by dfunckt