balena-io/etcher #1808: Etcher 1.1.x cannot scan the drives on Windows 10 x64, meaning it cannot flash any

Page-, 2017-10-24:
- **Etcher version:** 1.1.0, 1.1.1, 1.1.2
- **Operating system and architecture:** Windows 10 x64
- **Image flashed:** N/A
- **Do you see any meaningful error information in the DevTools?** 

On 1.1.0 I get
```
Error: Couldn't scan the drives: Invalid handle
```
followed by
```
Error: write EPROTO 101057795:error:140770FC:SSL routines:SSL23_GET_SERVER_HELLO:unknown protocol:c:\\jenkins\\workspace\\electron-win-x64\\vendor\\node\\deps\\openssl\\openssl\\ssl\\s23_clnt.c:794:\n\n    at exports._errnoException (util.js:1022:11)\n    at WriteWrap.afterWrite [as oncomplete] (net.js:804:14)
```

On 1.1.1 I get just the `Error: Couldn't scan the drives: Invalid handle` error.

On 1.1.2 I get
```
Error: Couldn't scan the drives: Permission error
```

1.0.0 works fine

jviotti, 2017-10-27:
Weird. Can you reproduce the error using the `drivelist` module directly?

```sh
git clone https://github.com/resin-io-modules/drivelist
cd drivelist
npm install
node example
```

If so, can you run it with the `DRIVELIST_DEBUG` environment variable set, which will reveal a lot more debugging messages?

```sh
set DRIVELIST_DEBUG=1
node example
```

Page-, 2017-10-27:
I had to use `$env:DRIVELIST_DEBUG=1` instead due to powershell (which is the default nowadays) and the result was:
```
[drivelist] Initializing scanner
[drivelist] Initializing COM
[drivelist] Initializing COM security levels
[drivelist] Creating connection instance
[drivelist] Connecting to server
[drivelist] Setting proxy security levels
[drivelist] Interpreting error 0x0
[drivelist] Getting list of drives from WMI
[drivelist] Getting id
[drivelist] Processing \\.\PHYSICALDRIVE0
[drivelist] Getting caption
[drivelist] Getting disk information
[drivelist] Opening handle on \\.\PHYSICALDRIVE0
[drivelist] Getting drive geometry
[drivelist] Getting system volume drive letter
[drivelist] Getting available volumes
[drivelist] Found volume: C
[drivelist] Getting volume type
[drivelist] Checking if volume has filesystem
[drivelist] Getting device number
[drivelist] Checking if disk is writable
[drivelist] Found volume: X
[drivelist] Getting volume type
[drivelist] Checking if volume has filesystem
[drivelist] Getting device number
[drivelist] Interpreting error 0x80070005
Error: Couldn't scan the drives: Permission error
```

Page-, 2017-10-27:
Just plugged in an sd card in case that makes a difference and got:
```
[drivelist] Initializing scanner
[drivelist] Initializing COM
[drivelist] Initializing COM security levels
[drivelist] Creating connection instance
[drivelist] Connecting to server
[drivelist] Setting proxy security levels
[drivelist] Interpreting error 0x0
[drivelist] Getting list of drives from WMI
[drivelist] Getting id
[drivelist] Processing \\.\PHYSICALDRIVE0
[drivelist] Getting caption
[drivelist] Getting disk information
[drivelist] Opening handle on \\.\PHYSICALDRIVE0
[drivelist] Getting drive geometry
[drivelist] Getting id
[drivelist] Processing \\.\PHYSICALDRIVE1
[drivelist] Getting caption
[drivelist] Getting disk information
[drivelist] Opening handle on \\.\PHYSICALDRIVE1
[drivelist] Getting drive geometry
[drivelist] Getting system volume drive letter
[drivelist] Getting available volumes
[drivelist] Found volume: C
[drivelist] Getting volume type
[drivelist] Checking if volume has filesystem
[drivelist] Getting device number
[drivelist] Checking if disk is writable
[drivelist] Found volume: D
[drivelist] Getting volume type
[drivelist] Checking if volume has filesystem
[drivelist] Getting device number
[drivelist] Checking if disk is writable
[drivelist] Found volume: E
[drivelist] Getting volume type
[drivelist] Checking if volume has filesystem
[drivelist] Getting device number
[drivelist] Checking if disk is writable
[drivelist] Found volume: F
[drivelist] Getting volume type
[drivelist] Checking if volume has filesystem
[drivelist] Getting device number
[drivelist] Checking if disk is writable
[drivelist] Found volume: G
[drivelist] Getting volume type
[drivelist] Checking if volume has filesystem
[drivelist] Getting device number
[drivelist] Checking if disk is writable
[drivelist] Found volume: H
[drivelist] Getting volume type
[drivelist] Checking if volume has filesystem
[drivelist] Getting device number
[drivelist] Checking if disk is writable
[drivelist] Found volume: X
[drivelist] Getting volume type
[drivelist] Checking if volume has filesystem
[drivelist] Getting device number
[drivelist] Interpreting error 0x80070005
Error: Couldn't scan the drives: Permission error

jviotti, 2017-10-27:
Awesome, thanks for the logs @Page-. One more question: what is drive X? The code always seem to fail there, so it might be a kind of drive we're not accounting for. Is it an encrypted drive by any chance?

NeoMod, 2017-10-29:
Hello,

I've noticed the same problem, and I don't know if It can be useful but I have two BitLocker Encrypted drives in my system. (although none of it it's using the "X" drive letter)

Page-, 2017-10-30:
@jviotti, it's just a fake drive created by [SUBST](https://technet.microsoft.com/en-us/library/bb491006.aspx) pointing to a path on C:/ for convenience

jviotti, 2017-10-30:
Interesting! @jhermsmeier you might want to check and fix these edge cases ^

jhermsmeier, 2017-10-30:
@Page- oh, that's a known issue: https://github.com/resin-io-modules/drivelist/issues/158
@NeoMod the issue with BitLocker volumes is also known: https://github.com/resin-io/etcher/issues/1693

joem5636, 2017-11-12:
I started Etched in Admin mode and it worked???????

covkid50, 2017-11-21:
I also have the same problem. I am running Windows 10 64 bit, latest version, and Etcher 1.1.2. I get the error message "Couldn't scan the drives: permission error". After selecting an image, I cannot select a drive. I get the message "Connect a drive. No removable drive detected". I tried running in Admin mode, but with the same result. Strangely. on another PC running exactly the same versions of Windows and Etcher, I have no issue. I am not using the SUBST command, or BitLocker. I am afraid I don't know how to create an error log to post here. I'm not a programmer, just somebody wanting to use Etcher to burn an image to a USB drive. Any suggestions would be appreciated.

joem5636, 2017-11-22:
Works now without admin for some reason???

sevaseva2001, 2017-12-13:
the same problem.
Windows 10 64 bit, latest version, and Etcher-Portable-1.2.1-x64
and then "Couldn't scan the drives: permission error". and  "Connect a drive. No removable drive detected"
Why?

patric-r, 2017-12-23:
Same error here (windows 10 x64). Using debug procedure stated above, I was able to identify the problematic drive letter (in my case, Q:). Interestingly, windows drive management doesn't show up this drive and I don't know yet where this drive letter comes from.
However, i was able to dismount it, using `mountvol Q:  /D`  and afterwards, etcher worked seamlessly.

covkid50, 2018-01-19:
> This has been fixed in #1953

Thanks!

dhawken, 2019-02-13:
I see this issue is resolved, but i just wanted to report that I was still getting some weird behaviour similar to this issue.  I had an SD card plugged into a reader on Win 10 x64 and got the "no scannable drives" issue.  However, when i removed the SD card, all drives appeared.  Then when i plugged the SD card back in, that drive also added to the list.  Possible workaround for anyone who finds this issue.  I did not see any error logs in dev tools on initial load.