balena-io/etcher #1809: Don't open another Etcher instance when opening it again

jhermsmeier, 2017-10-25:
When you click on Etcher and the program is running, it opens a new instance of Etcher. Instead it should activate / focus the currently running Etcher instance.

jviotti, 2017-10-25:
I don't think this is the case of macOS, but keep in mind that we know of some users opening multiple instances as a way to do multi-writes, so we might want to delay this until proper multi-writes are implemented.

aw5m2nite, 2017-11-02:
It's with Windows 10.  When clicking on Etcher in the taskbar (already opened), it doesn't switch windows to Etcher, but insteaad opens a new instance of Etcher.

lurch, 2017-11-07:
I wonder if this an Electron / Windows 10 issue, rather than specifically an Etcher issue? \*shrug\*

@aw5m2nite Is this with the 64-bit Windows installer version of Etcher 1.1.2 ?

Fireflaker, 2018-03-09:
Same issue. It had worked. Not anymore. Uninstalled and re-installed. @Win10x64

closed 2018-10-15 by jviotti