balena-io/etcher #1810: Replace Lodash templates

Shou, 2017-10-25:
Instead we could simply use lambdas and native template literals, e.g.

```js
const tplt = _.template('I am a <%= adjective %> template.')
tplt({ adjective: 'small' })

const tplt = ({ adjective }) => `I am a ${adjective} template.`
tplt({ adjective: 'small' })
```

lurch, 2017-10-25:
Would this change have any knock-on effects if/when we decide to localize Etcher? ( #308 )

Shou, 2017-10-25:
Can't see how, I don't believe `_.template` has any functionality that localization could leverage.

lurch, 2017-10-25:
Fair enough :)  I just wondered if searching for `_.template` might be a useful way to find strings that ought to be localised, but thinking about it a bit more I guess there'll also be plenty of strings that we need to localise that _aren't_ templated.

jhermsmeier, 2017-10-25:
In terms of localization, string template could be easily fitted with it by sticking a template function in front of it (see [tagged template literals](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Template_literals#Tagged_template_literals)), far easier than mending lodash templates.

urohit011, 2017-10-30:
I would like to work on this.

commit df2ebf93 (2018-01-29): refactor: replace lodash templates #1810 (#2006)

We replace the lodash templates with arrow-functions and change the
single-argument object into multiple arguments.

Fixes #1810 
Closes #2006 

Change-Type: patch
Changelog-Entry: Replace Lodash templates with arrow-functions.
closed 2018-01-29 by jhermsmeier