balena-io/etcher #1812: macOS Sierra Verification Fails For ANY image

niftylettuce, 2017-10-27:
Every image I've tried fails.  This is on a brand new SDCARD.

[image]

Tested Ubuntu Raspberry Pi 3 image as well as Ubuntu Mate Raspberry Pi 3 image.

https://wiki.ubuntu.com/ARM/RaspberryPi
https://ubuntu-mate.org/raspberry-pi/

niftylettuce, 2017-10-27:
Using SanDISK Class 10 32, 64, 128, 256 GB cards all failed with same error

lurch, 2017-10-27:
What SD card reader are you using? Where did you buy your SD cards from? Is it possible you've bought a bulk-set of 'fake' SD cards?

niftylettuce, 2017-10-27:
I believe this is an error because Mac cannot read ext4 and other formats.
The PI_ROOT partition doesn't show, only PI_BOOT.  I don't think Mac can
fix a SD card without using something like Gparted boot disk.

lurch, 2017-10-28:
> I believe this is an error because Mac cannot read ext4 and other formats.

That's not a problem for Etcher - it just reads and writes low-level bytes, so it doesn't matter what file-system they happen to be. (Etcher's verification process works just fine on Windows, which also can't read ext4 filesystems)

>  I don't think Mac can fix a SD card without using something like Gparted boot disk.

If you want to be able to 'repair' your SD cards so that they're usable with other programs, see https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#os-x

If you could please answer my earlier questions, we might be able to help you narrow down where the problem is :)

niftylettuce, 2017-10-29:
I bought from reputable SanDisk reseller, brand new unsealed packaging.  I am using built-in Macbook SD card reader.  Also, I went ahead and tested this on Ubuntu and had zero issues, verification passed as well.

closed 2017-10-29 by niftylettuce

lurch, 2017-10-30:
@jhermsmeier Do you think we should re-open this, or is this likely to be one of those rare one-off issues that we can't really do anything about?

hoangphan, 2018-03-04:
it seems i faced the same problem with Mac sierra (10.13.3) and any etcher version.
i got exactly the same result whenever trying to flash a raspbian to my SD card.

if anyone have success on overcoming this issue please share some information. Thanks!

jhermsmeier, 2018-03-04:
As @lurch already touched on, more often than not persistent verification failure leads back to the hardware (i.e. the sd-card reader, or the cards themselves) - is your mac connected to power while flashing?Are you using an external sd-card reader? And is it only Raspbian specifically, or does this happen with other images as well?

hoangphan, 2018-03-05:
i have mac connected to power while flashing, and also the case without power.
yes, i have both types of adapter for SD card of my micro SD 16GB, and the USB adapter type. both not work as the same result.

anyway, i had overcome this by an alternative: using another Windows PC and etcher can complete the validation after flashing.

an interesting point is, after the above mentioned error by @niftylettuce, the card seems not able to be mounted/ unmounted by Mac anymore, after many try, i ended up with using my CANON G16 camera to format it, and surprisingly, a camera can format the card when none of my PC (both windows and mac) can do, such a shame to apple and microsoft.

lurch, 2018-03-05:
Is it possible that this is the same issue I described [here](https://github.com/resin-io/etcher/issues/2085#issuecomment-368819762), i.e. the host OS auto-mounting the partitions on the SD card (after writing, but before verification finishes), which then causes the filesystem meta-data to change, which then causes Etcher's verification process to fail? 

\*shrug\* Or maybe I'm just clutching at straws... ;-)

hoangphan, 2018-03-05:
I think so :) it seem Apple wants to force every body to start using the MBP with touch pad, without SD card and all type C USB support.

raydot, 2018-04-02:
I get that error on PC.  On Mac with three different SD cards I get the "Please ensure you have necessary permissions to access this resource..." error.  

/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 {"stack":"Error: EACCES: permission denied, open '/dev/rdisk1'","message":"EACCES: permission denied, open '/dev/rdisk1'","code":"EACCES"}

The disk created on the PC has RASPPI desktop files loaded on to it and boots with the logo but then turns into the black screen of death.  Any thoughts?

zen, 2018-04-10:
Those symptoms are identical when using write protected card ;)

raydot, 2018-04-12:
@zen did you maybe miss the part where the card has the desktop files loaded onto it?  ;)

I got Etcher to work with yet a third PC and that card.  Between that and any number of other things I've seen I think it must be a hardware issue.

rlueder, 2018-06-26:
I think I'm experiencing a similar problem, tried burning a few different ISO to a 32GB Sandisk Cruzer, validation seems to to progress close to 90% done and then the message below appears:

[image]

lurch, 2018-06-26:
@rlueder Duplicate of #1428 ?

rlueder, 2018-06-29:
Yep, that sounds about right! thanks for the heads up 👍