balena-io/etcher #1819: Use PolKit for Wayland support

Shou, 2017-10-27:
With Ubuntu 17.10, Wayland is the default display server, and by default Etcher doesn't work.

The error  message (with gksudo):
[image]

```sh
Running gksudo                                                                                                     
Error copying '/run/user/1000/gdm/Xauthority' to '/tmp/libgksu-ibBRqB': No such file or directory  
```

[Issue on Redhat bugzilla closed as WONTFIX][1]:
> OK, to avoid the potential for misunderstood expectations: there are currently no plans to support running graphical apps with sudo under Wayland, and it seems quite unlikely that this will change anytime soon, so I'm going to close this as WONTFIX. The admin:// protocol mentioned by Ray is the way forward for applications that use gvfs to edit root-owned files. **Other applications need to use polkit to move privileged operations out of process.** This should really have happened many years ago, as polkit is no longer some new or mysterious technology.

[Insecure] workaround:
```sh
xhost +si:localuser:root
sudo etcher-electron
```

[1]: https://bugzilla.redhat.com/show_bug.cgi?id=1274451

jviotti, 2017-10-27:
Uh, the WONTFIX part sucks. How are we supposed to support polkit without requiring users to copy a certain configuration file into a certain part of the system? That obviously works for RPMs and DEBs, since we can add it to the package, but not for AppImages. 

Given USB is the only thing that needs this, can we put more time into documenting clear steps to allow a normal user to open USB devices, and then catching access errors on `node-usb`'s `.open()` and showing a nice user error documenting what the user needs to do? And then we can get rid of the elevation at startup Linux hack.

Shou, 2017-10-31:
It's not super pretty but could we run a stat for the config on startup and copy if it doesn't exist before spawning a process for the graphical parts?

Shou, 2017-11-02:
So with PolKit we're going to need a service running as root, where I assume we'll just re-use the child writer. Instead we could get around the whole thing by just making the child writer persistent like a service, a change that is platform agnostic so we don't have to maintain separate mechanisms – PolKit over DBus on Linux, IPC on other systems. Either way this ties heavily into the IPC channel stability.

lurch, 2017-11-07:
@Shou Does this mean that the current builds of Etcher AppImages don't work on Ubuntu 17.10 ?
If that's the case, then should we try to fix this before we publish Etcher 1.2.0 @jhermsmeier , otherwise we're likely to get loads of support requests / complaints?

And as a side-note, I'd also like to see:

> And then we can get rid of the elevation at startup Linux hack.

@Shou Can you remind me which distros, and in which situations, this "always run as root" on Linux behaviour is required? (FWIW I've never had any problems on Ubuntu 14.04)

jhermsmeier, 2018-01-18:
@Shou can this be closed? We're using polkit atm as far as I can see (?)

lurch, 2018-01-19:
I thought this bug would need to be revisited when we add usbboot support back into the Linux versions of Etcher, if we want that support to also work on Ubuntu 17.10? (and any other distros using Wayland, of course)

Shou, 2018-01-22:
Yeah, when we start adding persistence to the child writer we'll need to think about this again so this can stay open for now

closed 2018-10-15 by jviotti