balena-io/etcher #1823: Implement a new file / source picker

jhermsmeier, 2017-10-31:
Instead of a file picker, have a modal with 3 options:
  - type url
  - duplicate whole source
  - select file from source

If "select file" is preferred, show a very simple HTML file picker.
Also investigate the need for an on-screen keyboard for typing URLs (add to resin-electronjs?)

Shou, 2017-11-03:
Instead of a modal should we make it a page and move towards the V2 design? This also comes with the added benefit of allowing me to write it in React so we don't get more Angular into the codebase.

@konmouz could we get a mockup of the image selection screen with URL inputs and drive selection? We could potentially reuse the drive modal here if needed and just have a "select drive" button, if you feel like that's better. Also based on the V2 design I assume we want the recent selections in there too.

konmouz, 2017-11-07:
@Shou I'll take care of that within this or next week.

Shou, 2017-11-07:
Awesome.

lurch, 2017-11-07:
Sounds like this'd be a great place for closer interaction with https://github.com/resin-io-modules/resin-components ?

Shou, 2017-11-14:
@konmouz what's the status on this now?

konmouz, 2017-11-14:
@Shou I can push this and deliver today

jhermsmeier, 2018-03-05:
Duplicate of #950

curcuz, 2018-03-27:
ping @konmouz - any news?

konmouz, 2018-04-05:
@curcuz I need to redesign this and add some features but should be ready soon

closed 2018-10-15 by jviotti