balena-io/etcher #1828: Formatting flash drive afterwards

taranvohra, 2017-11-05:
- **Etcher version:**
1.1.2
- **Operating system and architecture:**
Windows 7 64-bit, Mac OSX 10.13 64-bit
- **Image flashed:**
Ubuntu 14.04
- **Do you see any meaningful error information in the DevTools?**
No

So I flashed an ISO of ubuntu14.04. After using it, I noticed I cannot go back and format my flash drive back to its original form. It says 18KB free of ~ 2MB. What is the solution to this?
[image]

jhermsmeier, 2017-11-05:
This happens when Windows can only read one or few partitions due to the filesystems in the image (which is perfectly normal) – you should be able to erase any volumes on that drive through Disk Management and re-format it, or if Windows won't let you do that, see our section on [recovering "broken" drives](https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives) in the documentation for details on other ways.

closed 2017-11-05 by jhermsmeier

xcursex, 2020-04-13:
Found this fix online:

Sometimes, things might go wrong, and you end up with a half-flashed drive that is unusable by your operating systems, and common graphical tools might even refuse to get it back to a normal state.

To solve these kinds of problems, we've collected a list of fail-proof methods to completely erase your drive in major operating systems.

Windows

In Windows, we'll use diskpart, a command line utility tool that comes pre-installed in all modern Windows versions.

Open cmd.exe from either the list of all installed applications, or from the "Run..." dialog usually accessible by pressing Ctrl+X.

Type diskpart.exe and press "Enter". You'll be asked to provide administrator permissions, and a new prompt window will appear. The following commands should be run in the new window.

Run list disk to list the available drives. Take note of the number id that identifies the drive you want to clean.

Run select disk N, where N corresponds to the id from the previous step.

Run clean. This command will completely clean your drive by erasing any existent filesystem.

OS X

Run the following command in Terminal.app, replacing N by the corresponding disk number, which you can find by running diskutil list:

diskutil eraseDisk free UNTITLED /dev/diskN
GNU/Linux

Make sure the drive is unmounted (umount /dev/xxx), and run the following command as root, replacing xxx by your actual device path:

dd if=/dev/zero of=/dev/xxx bs=512 count=1 conv=notrunc

71m024, 2025-02-06:
I use OSX and only this command did the Trick: `sudo dd if=/dev/zero of=/dev/xxx bs=512 count=1 conv=notrunc`