balena-io/etcher #1837: Corrupted drive after writing

Ben1190, 2017-11-10:
- **Etcher version:**
- **Operating system and architecture: Mac OS  El Capitan
- **Image flashed: Linux mint 18.2 cinnamon desktop

I was flashing the Linux mint cinnamon desktop 18.2 to a 32 gb drive and after the image finnished flashing, my computer showed the disk was unreadable and after I ran disk repair the drive showed it had a capacity of 3 megabytes.

Ben1190, 2017-11-10:
The version of etcher is the latest version.

lurch, 2017-11-10:
This is to be expected, because MacOS can't read the ext4 filesystem used by Linux disk images.
To 'repair' your disk, see https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#os-x

closed 2017-11-10 by lurch

Ben1190, 2017-11-10:
Since MacOS isn't compatible with the ext4 filesystem, Is it possible to flash my drive through a Linux virtual machine?

lurch, 2017-11-11:
MacOS _itself_ can't natively read ext4 filesystems, but Etcher on MacOS can write (disk images containing) ext4 filesystems no problem. Etcher just copies the disk image byte-by-byte onto the destination drive/SD card, so as far as Etcher is concerned it doesn't matter _what_ filesystems are in the disk image, or what OS you're using to do the flashing.

As long as Etcher's verification stage reaches 100%, you can be sure that your disk image was correctly written.