balena-io/etcher #1839: FreeBSD images crash windows.. /suggestion

ScotianOne, 2017-11-10:
- **Etcher version:** v1.1.2
- **Operating system and architecture:** Windows 7 Professional 64bit
- **Image flashed:** FreeBSD-11.1-RELEASE-i386-memstick.img
- **Do you see any meaningful error information in the DevTools?** No..

Flashed my img with no issues until the very end, eject on success.. which promptly crashed my system hard (bluescreen).

I went on to install FreeBSD with no issues however upon inserting the stick back into the win7 system it crashes the system every single time instantly. Took it to a Win10 system I had on standby and no problem there at all. Seems to be a common problem: https://forums.freebsd.org/threads/62230/

I've since resolved my issue and managed to write another image, however it may be useful if Etcher can be forced to make use of the entire stick via an option?  (rather than create a small partition) An average person without an army of computers at their disposal might not be able to get past the issue..

lurch, 2017-11-11:
Hmmm, annoying that a misplaced secondary GPT causes Windows to crash, what a horrible bug :-(  However this is obviously a problem of Windows and the FreeBD images, not an Etcher problem. The whole "modus operandi" of Etcher is to just copy whatever disk-image it's been given byte-for-byte onto the destination drive, without modifying the image at all.

I'll leave this issue open though, as @jhermsmeier has some familiarity with GPT images, so he might have some useful suggestions... ? (ping @zvin too in case he has any ideas)

jhermsmeier, 2017-11-14:
Oh, this is indeed very interesting! I've never had Windows BSOD on me with malformed / misplaced GPTs so far. I'll have a look at the image some time to see what's actually going on there :)

jhermsmeier, 2018-01-16:
Closing, as this is an issue with the FreeBSD image itself (according to https://forums.freebsd.org/threads/62230/ the secondary GUID Partition Table in the image is faulty) – which is something Etcher can't fix.

closed 2018-01-16 by jhermsmeier