balena-io/etcher #1840: Etcher will not go full screen

JBoyAdvance, 2017-11-11:
- 1.1.2
- Mac OSX 64 bit
- Raspbian
- No

It will not resize to full screen when you click the + button.

jezmck, 2017-11-11:
The maximise button in Windows is greyed out, so I guess the interface isn't designed for it.

Any particular reason you want it full screen?

JBoyAdvance, 2017-11-12:
No, there is not a specific reason. I just was wondering why it did not go full screen. It'd be nice if it did, though.

lurch, 2017-11-12:
Correct. The Etcher GUI is currently non-resizable, and therefore can't be maximised. On Linux the maximise button isn't shown either, so perhaps the maximise button getting shown on OSX is a bug in Electron? Or maybe OSX simply doesn't allow the maximise button to be disabled / removed? \*shrug\*

closed 2017-12-04 by jviotti