balena-io/etcher #1844: Etcher thinks images are too small when they are not

ladyada, 2017-11-14:
-  v1.0.0-beta19
- Win 7 x64 pro
- custom raspbian
- no error in devtools

Hiya, I have a custom image that is 3,965,190,144 bytes 

[image]

and trying to burn to 4G card which reports 3,955,228,672 bytes free. 
[image]

Yet etcher says there's not enough space

[image]

win32 disk imager works fine - some math calculation that is not working out?

jhermsmeier, 2017-11-14:
I'm confused – Etcher is doing the right thing here, isn't it? The 4 GB card is a whole 9.5 MB smaller than the image if I read your numbers correctly.

ladyada, 2017-11-14:
yes! sorry, etcher is correct, but...win32 imager writes it fine...which makes me wonder if there's some other lower-level-space-calculation that is not related to what the OS reports!

jhermsmeier, 2017-11-14:
No, but there might be a difference in what Win32DiskImager sees if it's using a different Windows API than we are to get the device size – there are some Win APIs (the COM stuff, for example) that still report it based on CHS addressing, where the size cannot be represented precisely for a lot of storage devices.

ladyada, 2017-11-14:
hmm yah - the reason this comes up for me is that if you 'read' the SD card using win32, it creates an image that is that size. so basically you cannot use etcher to burn images that are backups. 

anyways, its not a huge deal!

jhermsmeier, 2017-11-14:
Ooh... I didn't realise that at first; but in this case we should look into it – not that we're screwing something up here. I'll try to reproduce this.

ladyada, 2017-11-14:
np - sorry initial report was super unclear :D

jhermsmeier, 2017-11-14:
Because I am an idiot, and can't read, I didn't see that you're still on Etcher v1.0.0-beta19 – that is a version that's using an older version of drivelist that *does still use the inaccurate CHS APIs*. Can I ask you to try this again with the current Etcher v1.1.2?

lurch, 2017-11-14:
Hi @ladyada I've just noticed that in your first post you said you're using Etcher v1.0.0-beta19 ? That's a really old version, and it suffers from "using the COM API which CHS addressing to calculate the disk-size" bug https://github.com/resin-io-modules/drivelist/issues/142
If you use the latest version of Etcher (v1.1.2) you should find that Etcher is able to write your disk-image without any problem :)

The reason that Windows shows the disk as having a capacity of 3,955,228,672 bytes is because this is the capacity of the _filesystem_, not of the raw disk.

jhermsmeier, 2017-11-14:
I've just tried to reproduce this with Etcher v1.1.2, and there's no size difference between Win32DiskImager and Etcher, so that should work.

ladyada, 2017-11-14:
omg wait, i always update when i get a popup - but yeah now that i think of it i haven't seen an update in a while. *sigh* ok thanks i will manually update!

ladyada, 2017-11-14:
yep all good now. thank you!

closed 2017-11-14 by ladyada

jhermsmeier, 2017-11-14:
@lurch there's a difference of 9.5 MB (!) between the two though – that seems a bit excessive – wonder where that's coming from.

jhermsmeier, 2017-11-14:
Good to hear @ladyada! Glad we don't have a bug here after all :)

lurch, 2017-11-14:
> there's a difference of 9.5 MB (!) between the two though

Perhaps whatever formatting tool is being used didn't create the partition spanning the whole extent of the disk?