balena-io/etcher #1845: Visual - More info on file and device selected (too much truncated)

privacy-wolf, 2017-11-15:
Information about which iso-file and which usb-device was actually selected is quite important - the information in the Etcher screen is rather short (truncated) which makes it sometimes hard to read/recognize it - although there would be enough space to provide more information.

I would suggest to move the size information (for filename and device) on a separate line just below the file and device name thus giving more space for longer names (cf. screenshot below).

[image]

lurch, 2017-11-15:
If you mouse-hover over the truncated names you can see more info - is that what you're looking for?
We try very hard to keep the Etcher GUI as "clean" and as user-friendly as possible, without bombarding the user with information that they may not need.

alexandrosm, 2017-11-20:
@lurch I think we can afford to put the size on the next line though. @konmouz what do you think?

konmouz, 2017-11-20:
@lurch @alexandrosm, we can do that, the only small issue is that we also have the 'change' CTA underneath. To avoid making it look like 3 lines of text, we could give a good amount of vertical space (e.g. 2 lines) between 'image size' and 'change'

alexandrosm, 2017-11-23:
worth trying

lurch, 2018-02-28:
Duplicate of #1292

closed 2018-02-28 by lurch