balena-io/etcher #1860: Escalation issue

craig-mulligan, 2017-11-17:
- **Etcher version:** 1.2.0 (unreleased) but also seen on previous versions.
- **Operating system and architecture:** macOs 10.12.6 (16G29)
- **Image flashed:** resin image
- **Do you see any meaningful error information in the DevTools?**

```
Fri Nov 17 2017 18:40:35 GMT+0000 (GMT) Elevation cancelled ({"image":"/Users/gaudi/Downloads/resin.img","drive":{"size":7969177600,"raw":"/dev/rdisk2","displayName":"/dev/disk2","protected":false,"system":false,"adapter":"standard","device":"/dev/disk2","mountpoints":[],"description":"STORAGE DEVICE"},"uuid":"7b451054-d855-434d-bb11-71b2fd5e9788","unmountOnSuccess":false,"validateWriteOnSuccess":false})
```

I've tried multiple images/drives. I noticed the problem after I added windows with bootcamp. May just be coincidence but seems relevant.

jhermsmeier, 2017-11-17:
This is very relevant, thank you! Can you dump the output of `node -v`, `npm -v`, and `npm ls` as well as some more context around the "Elevation cancelled" message here? That'd be very helpful :)

craig-mulligan, 2017-11-17:
```
➜  etcher git:(master) node -v
v6.11.3
➜  etcher git:(master) npm -v 
3.10.10
➜  etcher git:(master) npm ls
etcher@1.1.2 /Users/gaudi/etcher
├── angular@1.6.3
├─┬ angular-if-state@1.0.0
│ └── angular-ui-router@0.3.2
├── angular-middle-ellipses@1.0.0
├── angular-mocks@1.6.3
├─┬ angular-moment@1.0.1
│ └── moment@2.13.0
├── angular-seconds-to-date@1.0.0
├── angular-ui-bootstrap@2.5.0
├── angular-ui-router@0.4.2
├─┬ asar@0.10.0
│ ├── chromium-pickle-js@0.1.0
│ ├─┬ commander@2.9.0
│ │ └── graceful-readlink@1.0.1
│ ├── cuint@0.2.2
│ ├─┬ glob@6.0.4
│ │ ├─┬ inflight@1.0.6
│ │ │ └── wrappy@1.0.2
│ │ ├── inherits@2.0.1
│ │ └── once@1.3.3
│ ├─┬ minimatch@3.0.3
│ │ └─┬ brace-expansion@1.1.6
│ │   ├── balanced-match@0.4.2
│ │   └── concat-map@0.0.1
│ ├─┬ mkdirp@0.5.1
│ │ └── minimist@0.0.8
│ └─┬ mksnapshot@0.1.0
│   ├─┬ decompress-zip@0.1.0
│   │ ├─┬ binary@0.3.0
│   │ │ ├── buffers@0.1.1
│   │ │ └─┬ chainsaw@0.1.0
│   │ │   └── traverse@0.3.9
│   │ ├─┬ graceful-fs@3.0.11
│   │ │ └── natives@1.1.0
│   │ ├── mkpath@0.1.0
│   │ ├── q@1.4.1
│   │ ├─┬ readable-stream@1.1.14
│   │ │ └── isarray@0.0.1
│   │ └─┬ touch@0.0.3
│   │   └── nopt@1.0.10
│   ├─┬ fs-extra@0.18.2
│   │ └── graceful-fs@3.0.11
│   └─┬ request@2.55.0
│     ├── aws-sign2@0.5.0
│     ├─┬ bl@0.9.5
│     │ └── readable-stream@1.0.34
│     ├── caseless@0.9.0
│     ├─┬ combined-stream@0.0.7
│     │ └── delayed-stream@0.0.5
│     ├─┬ form-data@0.2.0
│     │ └── async@0.9.2
│     ├─┬ har-validator@1.8.0
│     │ ├── bluebird@2.11.0
│     │ └── commander@2.9.0
│     ├── hawk@2.3.1
│     ├─┬ http-signature@0.10.1
│     │ ├── asn1@0.1.11
│     │ ├── assert-plus@0.1.5
│     │ └── ctype@0.5.3
│     ├─┬ mime-types@2.0.14
│     │ └── mime-db@1.12.0
│     ├── node-uuid@1.4.7
│     ├── oauth-sign@0.6.0
[... 1628 lines cut ...]
│ │ ├─┬ load-json-file@1.1.0
│ │ │ ├── graceful-fs@4.1.4
│ │ │ ├─┬ parse-json@2.2.0
│ │ │ │ └─┬ error-ex@1.3.0
│ │ │ │   └── is-arrayish@0.2.1
│ │ │ └── strip-bom@2.0.0
│ │ └── symbol@0.2.3
│ ├─┬ read-pkg-up@1.0.1
│ │ └─┬ read-pkg@1.1.0
│ │   └─┬ path-type@1.1.0
│ │     └── graceful-fs@4.1.4
│ ├── require-main-filename@1.0.1
│ ├── set-blocking@1.0.0
│ ├─┬ string-width@1.0.1
│ │ ├─┬ code-point-at@1.0.0
│ │ │ └── number-is-nan@1.0.0
│ │ └── is-fullwidth-code-point@1.0.0
│ ├── window-size@0.2.0
│ ├── y18n@3.2.1
│ └─┬ yargs-parser@2.4.0
│   └── camelcase@2.1.1
└─┬ yauzl@2.6.0
  └─┬ fd-slicer@1.0.1
    └── pend@1.2.0
```

jhermsmeier, 2017-11-17:
I've just pushed a `update-sudo-prompt` branch which updates `sudo-prompt`, if you'd like to try that – not sure what might be causing it atm, as I can't reproduce this :/

jviotti, 2017-11-17:
@craig-mulligan Did the updated sudo prompt work?

jviotti, 2017-11-17:
For the record, this only happens when this condition is true: https://github.com/jorangreef/sudo-prompt/blob/v6.1.0/index.js#L316

jviotti, 2017-11-17:
@craig-mulligan Do you get to see the elevation dialog?

craig-mulligan, 2017-11-20:
@jviotti no, I didn't get a prompt.

To ensure I'm testing correctly I followed the following. 
```
run `make electron-develop` on `update-sudo-prompt branch`
run `npm start` and try flash image.
See the same result with elevation cancelled a second after starting. 
```

jviotti, 2017-11-20:
@craig-mulligan You're doing it right :+1: I reviewed the `sudo-prompt` changes since the version we were using and I suspected the problem would persist.

Can you try running the hello world example `sudo-prompt` provides in the "Usage" section: https://github.com/jorangreef/sudo-prompt#usage?

craig-mulligan, 2017-11-20:
Hey @jviotti, I tried running 
```
sudo.exec('echo hello', {},
  function(error, stdout, stderr) {
    if (error) throw error;
    console.log('stdout: ' + stdout);
  }
);
```
And I get: 
```
/Users/gaudi/work/sudo-test/index.js:8
    if (error) throw error;
               ^

Error: User did not grant permission.
    at ReadFileContext.callback (/Users/gaudi/work/sudo-test/node_modules/sudo-prompt/index.js:316:49)
    at FSReqWrap.readFileAfterOpen [as oncomplete] (fs.js:367:13)
```

jviotti, 2017-11-20:
I opened an issue on `sudo-prompt`: https://github.com/jorangreef/sudo-prompt/issues/64

lurch, 2017-12-02:
Should this be removed from the v1.2.0 milestone, given that Etcher v1.2.0 has already been released?

jviotti, 2017-12-05:
I'll remove this from the v1.2.1 milestone as it seems it was an issue on @hobochild 's mac, and not something we can do anything about from the Etcher side.

I'll still leave it open to see if we can get to the bottom of it in https://github.com/jorangreef/sudo-prompt/issues/64, and hopefully provide some documentation.

lurch, 2017-12-06:
In [this comment](https://github.com/jorangreef/sudo-prompt/issues/64#issuecomment-349629880) @hobochild says that he reinstalled his macOS and the issue has now gone away. Which makes it look like this was a one-off glitch unrelated to Etcher, so I'll close this issue but we can re-open it if anyone else runs into it in future.

closed 2017-12-06 by lurch

jhermsmeier, 2017-12-06:
Where "one-off glitch" means a corrupted Mac OS install

lurch, 2017-12-06:
i know nothing about Mac OS, so feel free to re-open... :)

jhermsmeier, 2017-12-06:
Nah, all god; I just meant to clarify that this happens if your Mac OS install is broken, as detailed in https://github.com/jorangreef/sudo-prompt/issues/64