balena-io/etcher #1865: Stuck at 40% during flashing

RebelReign, 2017-11-18:
Trying to write this .img file to an SD card and I'm stuck at 40% in flashing as far as i can tell there are no errors that would be causing this.
- Etcher version: v1.1.2 - 2017-08-07
- Operating system and architecture: kali linux, x86_64
- Image flashed: 2016-05-10-raspbian-jessie-lite-video-looper.img

- Do you see any meaningful error information in the DevTools?
[-1511027911872.log](https://github.com/resin-io/etcher/files/1484864/-1511027911872.log)

ignore the dbg nonsense at the bottom that was me

jhermsmeier, 2017-11-18:
Duplicate of #1423

jhermsmeier, 2017-11-18:
Thanks for reporting! This has been fixed in https://github.com/resin-io/etcher/pull/1843 and will be included in the next release.

closed 2017-11-18 by jhermsmeier

RebelReign, 2017-11-18:
Thank you for the response!