balena-io/etcher #1866: Etcher waits for a thread. GUI not showing. Win10

hmendozap, 2017-11-19:
- **Etcher version: 1.1.2**
- **Operating system and architecture: Windows 10 - 64 bits**
- **Image flashed: No**
- **Do you see any meaningful error information in the DevTools? No, gui could not be started**

When starting nothing is displayed, but it shows in running processes. When analyzing wait threads, it appears that etcher is waiting for a thread in etcher. 

I had to restart the machine and use the portable version.

lurch, 2017-11-28:
This is definitely an unusual problem, which I don't think anybody has reported before. Can you think of anything "unusual" about your system setup that might explain why you're seeing this problem when we haven't? :-S  What CPU / RAM / etc. does your computer have?

hmendozap, 2017-12-08:
Unusual? I don't think so... I have a NVIDIA GPU GTX960M other than that your usual Win Developer environment: i7, 8GB RAM, VStudio 2017, Bash on windows, CUDA, Python... don't know if that helps.

Thanks for the work, guys. If I can be of service I'd gladly help

euh, 2017-12-15:
Same problem with v1.2.1 installer AND portable. Windows 7 64bits in my case.
I have no problem with v1.1.2 portable, I have not tested with the v1.2.1 installer.

lurch, 2017-12-15:
Could the people suffering from this problem also try starting Etcher from the Command Prompt, to see if that reveals any additional output?

euh, 2017-12-17:
From Command Prompt.. I see no output.

[image]

[image]

adipascu, 2018-01-06:
I have the same issue as @euh.
Etcher 1.3.0 is not showing the UI on windows 7 64 bit but the processes are running in the background.

1.1.2 works correctly.

megusd, 2018-01-06:
Same here. Running Win7 64
Where to get older versions>

adipascu, 2018-01-06:
@megusd You can find old releases here: https://github.com/resin-io/etcher/releases

lizzyfin, 2018-01-06:
Thank you so much for this tip. I had the same problem of the missing GUI on my Win7Pro-64 when trying the latest versions of etcher. The older 1.1.2 version worked just fine!

4biddenfruit, 2018-01-15:
Hey people, same problem or even more :/ as described latest version doesnt even shows a window both portable and setup, unfortunately 1.1.2 doesnt do the job either. Window stays blank white all the time also both options...

Now i need to search for an alternative :/ very disappointing.

My system:

Win 10 Pro
i5-4500k
16GB DDR4 RAM
GTX 960OC

Nionor, 2018-02-20:
Same problem here and I did some testing, all on the Portable versions.
1.3.1, 1.2.1 and 1.1.2 64bit versions wont start, ie. GUI missing.

1.3.1 32bit I get a GUI but with missing buttons 
[image]

1.2.1 and 1.1.2 32bit works ok

Acer laptop with
Win 10 Pro 64bit
i7-3632QM, 
16gig ram

jhermsmeier, 2018-02-21:
@Nionor can you open the DevTools in v1.3.1 (<kbd>Ctrl</kbd>+<kbd>Shift</kbd>+<kbd>I</kbd>) and post what appears in the console there?

JTT-AE, 2018-05-25:
Same problem on Win10, latest Etcher.

jhermsmeier, 2018-05-25:
@Nionor thanks – we'll dig into it!

matmai, 2018-06-10:
Same problem with Etcher-Portable-1.4.4-x64.exe running on Win 7 64bit.

toussa, 2018-06-22:
same problem here (no GUI at all, dangling process) with Etcher-Setup-1.4.4-x64.exe on Win10 64bit.
No output at all in the console.

GrandNewbien, 2018-07-13:
Same issue on latest version as well

ccjoness, 2018-07-18:
I am having the same problem.

matmai, 2018-07-18:
@jhermsmeier: Can you verify the problem or do you need any additional information? Is it a problem with [electron](http://electron.atom.io/)?

GrandNewbien, 2018-07-21:
@matmai I've seen this problem in other electron apps as well.

jhermsmeier, 2018-07-23:
@matmai we've not been able to successfully reproduce this so far (and thus also can't tell yet if it's Etcher- or Electron-specific), so any detailed info would help!

GrandNewbien, 2018-07-23:
https://github.com/electron-userland/electron-prebuilt-compile/issues/36

I've found this thread that helped solve the issue. I can confirm this issue only occurs on my acer laptop.

jhermsmeier, 2018-07-23:
@GrandNewbien nice find, thanks!

jhermsmeier, 2018-07-25:
@Ramzi-Sah do you also have an Acer laptop?

jhermsmeier, 2018-07-26:
@matmai what manufacturer is your computer from?

matmai, 2018-07-27:
@jhermsmeier It is a selfmade Desktop computer without any preinstalled software. I also tried to deactivate virus protection and [EMET](http://microsoft.com/emet) Tool without success.

D3faIt, 2018-08-15:
I also have this issue.
One thing to note, is that I have a Hashtag (#) in the username C:\Users\#Chiya\
This has caused a similar issue before with [Postman](https://www.getpostman.com/)

On Windows 8.1 64-Bit

afflicted, 2018-10-09:
Same problem with 1.4.4 on Windows 10 64bit running GTX960

Marooned-MB, 2018-10-13:
Windows 7 Pro 64 En
I've tried Etcher-Portable-1.4.5-x64 and every version down to 1.2.0. I've also tried x86. Always 3 processes are spawn (initially more) and no GUI is displayed. It's not working.

sbela, 2018-12-06:
Same here with 1.4.8-x64

thundron, 2019-01-19:
Are you using the latest v1.4.9 `balenaEtcher`?

GeoffWM, 2019-01-20:
I downloaded etcher from the balena site yesterday. I cannot ltell if it is not the latest as I cannot access the about menu.

GeoffWM, 2019-01-21:
It was download from the balena etcher website so it should be balenaetcher. Is there another way to determine which it is? I worked round it with Rufus, but balenaetcher shouldn't hang.

beerisgood, 2019-03-08:
Can confirm start problem with balenaEtcher-Portable-1.4.9-x64.exe under Windows 10

I got a error window with the message that at least Windows 7 is needed.

matmai, 2019-03-11:
Also current v1.5.5 release is still affected.

tmanthey, 2019-03-28:
Same thing happens on Linux Mint with 1.5.18

lurch, 2019-03-29:
@tmanthey I think that's a different problem - see #2715

TracesOfNut, 2019-04-16:
I am using Version 1.5.24.

It ran straight after installing, but after a reboot, it will only run when started in Administrator mode.

TracesOfNut, 2019-04-16:
I ran 1.5.24 in normal and administrator mode and logged with procmon.exe 

Just after etcher references electron.asar, access to users folder is denied.  After that a whole load of things start to close.  The following has the event time removed to make comparison easier.  

The one started in admin mode gets access to the user folder and continues ok.

```
balenaEtcher.exe,13356,CreateFile,C:\Users\Redacted\AppData\Local\Programs\balena-etcher\resources\electron.asar,SUCCESS,"Desired Access: Generic Read, Disposition: Open, Options: Synchronous IO Non-Alert, Non-Directory File, Attributes: n/a, ShareMode: Read, Write, AllocationSize: n/a, OpenResult: Opened"
balenaEtcher.exe,13356,ReadFile,C:\Users\Redacted\AppData\Local\Programs\balena-etcher\resources\electron.asar,SUCCESS,"Offset: 0, Length: 8, Priority: Normal"
balenaEtcher.exe,13356,ReadFile,C:\Users\Redacted\AppData\Local\Programs\balena-etcher\resources\electron.asar,SUCCESS,"Offset: 8, Length: 4,196"
balenaEtcher.exe,13356,CreateFile,C:\Users\Redacted\AppData\Local\Programs\balena-etcher\resources\electron.asar,SUCCESS,"Desired Access: Generic Read, Disposition: Open, Options: Synchronous IO Non-Alert, Non-Directory File, Attributes: n/a, ShareMode: Read, Write, AllocationSize: n/a, OpenResult: Opened"
balenaEtcher.exe,13356,ReadFile,C:\Users\Redacted\AppData\Local\Programs\balena-etcher\resources\electron.asar,SUCCESS,"Offset: 0, Length: 8, Priority: Normal"
balenaEtcher.exe,13356,ReadFile,C:\Users\Redacted\AppData\Local\Programs\balena-etcher\resources\electron.asar,SUCCESS,"Offset: 8, Length: 4,196"
balenaEtcher.exe,13356,CreateFile,C:\,SUCCESS,"Desired Access: Read Attributes, Synchronize, Disposition: Open, Options: Synchronous IO Non-Alert, Open Reparse Point, Attributes: n/a, ShareMode: Read, Write, Delete, AllocationSize: n/a, OpenResult: Opened"
balenaEtcher.exe,13356,QueryAllInformationFile,C:\,SUCCESS,"CreationTime: 26/11/2018 5:47:46 AM, LastAccessTime: 16/04/2019 10:44:12 AM, LastWriteTime: 16/04/2019 10:44:12 AM, ChangeTime: 16/04/2019 10:44:12 AM, FileAttributes: HSD, AllocationSize: 20,480, EndOfFile: 20,480, NumberOfLinks: 1, DeletePending: False, Directory: True, IndexNumber: 0x5000000000005, EaSize: 0, Access: Read Attributes, Synchronize, Position: 0, Mode: Synchronous IO Non-Alert, AlignmentRequirement: Long"
balenaEtcher.exe,13356,QueryInformationVolume,C:\,SUCCESS,"VolumeCreationTime: 14/04/2019 7:16:12 AM, VolumeSerialNumber: 0450-498F, SupportsObjects: True, VolumeLabel: "
balenaEtcher.exe,13356,CloseFile,C:\,SUCCESS,
balenaEtcher.exe,13356,CreateFile,C:\Users,SUCCESS,"Desired Access: Read Attributes, Synchronize, Disposition: Open, Options: Synchronous IO Non-Alert, Open Reparse Point, Attributes: n/a, ShareMode: Read, Write, Delete, AllocationSize: n/a, OpenResult: Opened"
balenaEtcher.exe,13356,QueryAllInformationFile,C:\Users,BUFFER OVERFLOW,"CreationTime: 15/09/2018 3:39:26 PM, LastAccessTime: 13/04/2019 2:09:21 PM, LastWriteTime: 5/04/2019 12:23:39 PM, ChangeTime: 16/04/2019 10:03:57 AM, FileAttributes: RD, AllocationSize: 4,096, EndOfFile: 4,096, NumberOfLinks: 1, DeletePending: False, Directory: True, IndexNumber: 0x1000000001495, EaSize: 0, Access: Read Attributes, Synchronize, Position: 0, Mode: Synchronous IO Non-Alert, AlignmentRequirement: Long"
balenaEtcher.exe,13356,QueryInformationVolume,C:\Users,SUCCESS,"VolumeCreationTime: 14/04/2019 7:16:12 AM, VolumeSerialNumber: 0450-498F, SupportsObjects: True, VolumeLabel: "
balenaEtcher.exe,13356,CloseFile,C:\Users,SUCCESS,

balenaEtcher.exe,13356,CreateFile,C:\Users\Redacted,ACCESS DENIED,"Desired Access: Read Attributes, Synchronize, Disposition: Open, Options: Synchronous IO Non-Alert, Open Reparse Point, Attributes: n/a, ShareMode: Read, Write, Delete, AllocationSize: n/a"

balenaEtcher.exe,13356,CloseFile,C:\Users\Redacted\AppData\Local\Programs\balena-etcher\resources\electron.asar,SUCCESS,
balenaEtcher.exe,13356,CloseFile,C:\Users\Redacted\AppData\Local\Programs\balena-etcher\resources\electron.asar,SUCCESS,
balenaEtcher.exe,13356,Thread Exit,,SUCCESS,"Thread ID: 14620, User Time: 0.0000000, Kernel Time: 0.0000000"
balenaEtcher.exe,13356,Thread Exit,,SUCCESS,"Thread ID: 15996, User Time: 0.0000000, Kernel Time: 0.0000000"
balenaEtcher.exe,13356,Thread Exit,,SUCCESS,"Thread ID: 22360, User Time: 0.0000000, Kernel Time: 0.0156250"
balenaEtcher.exe,13356,Thread Exit,,SUCCESS,"Thread ID: 25500, User Time: 0.0000000, Kernel Time: 0.0000000"
balenaEtcher.exe,13356,Thread Exit,,SUCCESS,"Thread ID: 27416, User Time: 0.0000000, Kernel Time: 0.0000000"
balenaEtcher.exe,13356,Thread Exit,,SUCCESS,"Thread ID: 19948, User Time: 0.0000000, Kernel Time: 0.0000000"
balenaEtcher.exe,13356,Thread Exit,,SUCCESS,"Thread ID: 21352, User Time: 0.0000000, Kernel Time: 0.0312500"
balenaEtcher.exe,13356,Thread Exit,,SUCCESS,"Thread ID: 7520, User Time: 0.0312500, Kernel Time: 0.0000000"
balenaEtcher.exe,13356,Thread Exit,,SUCCESS,"Thread ID: 19848, User Time: 0.0000000, Kernel Time: 0.0000000"
```

rjoomen, 2019-05-14:
Same here, Windows 10 1809, Etcher 1.5.38. Doesn't show a completed GUI, not even as admin, not even after a reboot.

thundron, 2019-05-14:
@rjoomen That's a different issue though, see https://github.com/balena-io/etcher/issues/2756 if that's what you see or open a new issue and fill out the information if it's not (and try the portable version as well)

Askill, 2019-06-28:
Same issue with latest version

pauldsmyth, 2020-01-29:
Same issue, version 1.5.72 gui never appears. Running from powershell returns:
PS C:\Users\paul> C:\Users\paul\AppData\Local\Programs\balena-etcher\balenaEtcher.exe
PS C:\Users\paul>
 Checking for update
Update for version 1.5.72 is not available (latest version: 1.5.72, downgrade is disallowed).

Just a blinking cursor, whether as a user or as an Administrator. Has a Windows update broke this?

BlueSaturn, 2020-02-03:
Same, I had to install the program.

areyoumadlad, 2020-02-05:
No GUI appearing with version 1.5.74 with either Portable or Install.
Same result after rebooting the system and trying to Run as administrator.

||| just a blank grey window |||

CTRL+SHIFT+I >>> _Developer Tools _Console

[error message in red]

=======================================

C:\Users\-_-\AppData\Local\Temp\1XMKJPoYGSl05MPXtmXo2X7Ea9d\resources\app.asar\node_modules\raven-js\dist\raven.js:58 {"stack":"TypeError: Cannot convert undefined or null to object\n    at Function.entries (<anonymous>)\n    at Object.init (file:///C:/Users/-_-/AppData/Local/Temp/1XMKJPoYGSl05MPXtmXo2X7Ea9d/resources/app.asar/generated/gui.js:3951:49)\n    at main (file:///C:/Users/-_-/AppData/Local/Temp/1XMKJPoYGSl05MPXtmXo2X7Ea9d/resources/app.asar/generated/gui.js:3633:12)","message":"Cannot convert undefined or null to object"}
console.<computed> @ C:\Users\-_-\AppData\Local\Temp\1XMKJPoYGSl05MPXtmXo2X7Ea9d\resources\app.asar\node_modules\raven-js\dist\raven.js:58

=======================================

msi GP63 Leopard 8RF [Laptop]
 ^ NVIDIA GeForce GTX 1070 + Intel UHD 630

Win 10.0.18363  [no updates available]
nothing unusual about system nor having any problems with anything else

trying an earlier version next

roy-t, 2020-02-05:
Same problem here, exactly the same error message as @areyoumadlad (`Cannot convert undefined or null to object`). I tried both the normal version and the portable version of 1.5.74.

Windows 10 64 bit v1809
NVIDIA Gefore RTX 2080

lurch, 2020-02-05:
For problems with 1.5.74 see #3062 (which I'm assuming will get merged fairly soon)

thundron, 2020-02-05:
Working on it, the CI is building the packages which takes ~1h so we can test the new release

matmai, 2020-02-08:
After Upgrading from Win7 to Win10 Etcher 1.5.5 now starts and shows its GUI.

thundron, 2020-02-10:
Solved in v1.5.75+

closed 2020-02-10 by thundron

lurch, 2020-02-10:
@thundron Has 1.5.75+ actually solved the problem that was _originally_ reported in this issue (which IIRC nobody at Balena has been able to reproduce yet?), or does it just fix the unrelated issues reported in the last week that were tacked onto the end of this thread? :wink: 
(if it's the latter, IMHO this shouldn't be closed yet :man_shrugging: )

thundron, 2020-02-12:
@lurch right.
This issue though refers to very old versions and many things have changed, some of them were also fundamental changes/version upgrades.
Last report was from 1.5.72 so before reopening, I think it's worth asking @pauldsmyth if he has had any luck with the latest versions, both Portable and Installer, and if not, if he can share a bit more wrt the behaviour

pauldsmyth, 2020-02-12:
I'll test again tomorrow. I gave up and went back to win32diskimager and
haven't needed to retest

pauldsmyth, 2020-02-26:
Sorry to be late in responding. Absolutely no change. Set up runs to install, progress box dissapears and that's it. 3 balenaSketcher processes showing in task manager but the gui never appears. Since you've closed this hastily without fully confirming the bug is fixed I'm out of here. I'll stick with win32diskimager.

pauldsmyth, 2020-02-26:
> @lurch right.
> This issue though refers to very old versions and many things have changed, some of them were also fundamental changes/version upgrades.
> Last report was from 1.5.72 so before [...]

Nope, 1.5.78 exactly the same behaviour

b-morgan, 2020-02-27:
I am the author of #3088 and would be happy to provide additional information or do additional testing / debugging either here or there.

donjakobo, 2020-05-14:
Can confirm, running Windows 10 too, Etcher version 1.5.88.0, same behaviour.  I see 3 processes running but nothing happens.  I tried running in Admin mode, but no change.

hmoffatt, 2020-05-27:
I just installed v1.5.93 on a fresh Windows 10 1809 x64 install and I have this problem.

Elliotsmith, 2020-05-27:
I also, appear to be getting this issue with v1.5.93.

b-morgan, 2020-05-27:
Apparently, the fix in #3160 didn't make the mainstream because this issue seems to indicate that the issue has returned in 1.5.93.

It would also be nice if the developers could tie all the issues that report the same problem together. I am aware of https://github.com/balena-io/etcher/issues/3088, https://github.com/balena-io/etcher/issues/3160#ref-pullrequest-622331470, and https://github.com/balena-io/etcher/pull/3165.

[26 more replies]