balena-io/etcher #1867: Couldn't scan the drives: An unknown error occurred (With a twist)

Arffeh, 2017-11-19:
- **Etcher version:**
1.1.2

- **Operating system and architecture:**
Win7 Ultimate 64 Bit

- **Image flashed:**
N/A

- **Do you see any meaningful error information in the DevTools?**

{"stack":"Error: Couldn't scan the drives: An unknown error occurred","message":"Couldn't scan the drives: An unknown error occurred"}

So, turns out Etcher spits the dummy if there is a bitlocker volume not yet decrypted.

Confirmed by having the volume decrypted, then proceeding to lock the volume again. the instant I do I get the error again.
Etcher needs to gracefully handle bitlocker volumes.
Was pulling my hair out trying to figure it out.

jhermsmeier, 2017-11-19:
Hey, this should've been fixed in #1692 and will ship in the next release

jhermsmeier, 2017-11-19:
Duplicate of #1687

closed 2017-11-19 by jhermsmeier