balena-io/etcher #1870: Add integration tests

jhermsmeier, 2017-11-20:
Possibly utilizing spectron

lurch, 2017-11-28:
I made a start on this a while ago, (IIRC it worked with both AppImages and .deb versions of Etcher) but I got stuck trying to figure out how to navigate the DOM
https://gist.github.com/lurch/6dac70712beef0a071cb02f4fd0bfa67

When I came back to this recently (for running smoke-tests on different Linux distros using Vagrant), I found that I had to manually downgrade the version of `webdriverio` for `spectron` to work correctly; but perhaps that's been fixed by now?

jviotti, 2017-11-28:
@lurch Great example! I'll send a quick PR setting up Spectron on the repo, so we can then start adding more and more tests there.

closed 2017-12-13 by jviotti