balena-io/etcher #1871: permission denied error

varunkrgupta03642, 2017-11-20:
Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Uno"

sketch\sketch_nov20a.ino.cpp:1:0: fatal error: can't open C:\Users\hp\AppData\Local\Temp\ccHsjdM0.s for writing: Permission denied

 

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

what to dp now

jhermsmeier, 2017-11-20:
I think you might have got the wrong repository here?

varunkrgupta03642, 2017-11-20:
i was trying to compiling my code when i was clicking verify button on my ardunio ide back to back it was compiling but when when i m clicking it and taking halt it was showing this kind of error

jhermsmeier, 2017-11-20:
Sorry, I'm confused, what does this have to do with Etcher? This is not an Arduino project, nor does it contain any Arduino related code (like the `sketch_nov20a.ino.cpp` that's mentioned in the error message).

closed 2017-11-28 by jhermsmeier