balena-io/etcher #1877: Request option to verify without flash

GitUser275, 2017-11-24:
- **Etcher version:**  1.2.0
- **Operating system and architecture:** Win 8.1 64 bit
- **Image flashed:**  Startux v1.4r3 (http://stratux.me/)
- **Do you see any meaningful error information in the DevTools?** No

Using new Sandisk 16 GB micro SD cards in the supplied SD card adapters I was unable to get a good verification after flash for several attempts.  I have eventual been able to get one of the cards to verify.

An older  8 GB micro SD card verifies ok about half the time.   I find it hard to believe that all these cards are defective and I'd like to be able to run multiple verification attempts without having to re-write the cards.   Does anyone else think that option would be useful?

Thanks.

lurch, 2017-11-28:
It might _sound_ like a useful option, but it'd be incredibly error-prone. Some filesystems store a last-mount-time (or last read time), which means _every_ time an SD card gets removed and inserted (and the operating system auto-mounts one of the partitions) then the data on the card changes, and so the verification would then fail, even though the user would believe that the data on the card hasn't changed.

@GitUser275 It sounds like maybe your SD card-writer is playing up? Do you have a different one you can try?

GitUser275, 2017-11-28:
I would not want to use the option after removing and inserting the card.   In the present implementation the user has no way to know if a verification error is caused by a write error or a read error.  There is no reason to write the card again if it was a read error.  All cards that fail verification  seem to work fine in the target system so I have no reason to believe the write failed.

It is possible the built in SD card slot of this laptop has an issue.   I prefer to use my Kodak USB multi card reader but Etcher does not find a card inserted in this device even though Explorer shows it and reads it.

jhermsmeier, 2017-11-28:
> There is no reason to write the card again if it was a read error.

Etcher has two steps; 1) flashing, and 2) verifying, which are also distinct in the UI; Write errors will cause 1) to bail out, while verification errors will happen during 2).
I guess we could retry reads when potentially temporary read errors happen during verification, and detail the error messages a bit better, to help distinguish between read/write errors.

> All cards that fail verification seem to work fine in the target system so I have no reason to believe the write failed.

That doesn't necessarily mean they're okay though; there are many things that can go wrong in places the system doesn't use at the time – from corruption in drivers to rarely or unused file system blocks. A lot of the time these errors will only become an issue at a later time.

lurch, 2017-11-28:
> I prefer to use my Kodak USB multi card reader but Etcher does not find a card inserted in this device even though Explorer shows it and reads it.

Did you try enabling "Unsafe Mode" in Etcher's settings?

GitUser275, 2017-11-28:
"Did you try enabling "Unsafe Mode" in Etcher's settings?"

No.  Why would I ever want to select an option that is named "unsafe"?  If you think that option would make a card in the Kodak reader available to Etcher I'll give it a try.

GitUser275, 2017-11-28:
I'm now suspecting that the Kodak reader is not SDHC compatible.  It's at least 10 years old.  I've been using it with 1 GB cards but, until this project, I don't think I have used it for anything larger.  I'll do some digging and report back.

GitUser275, 2017-11-28:
Yes, my Kodak card reader is specified for SD with no mention of SDHC.   Looks like I'll have to buy a more modern one.

lurch, 2017-11-29:
> No. Why would I ever want to select an option that is named "unsafe"?

Sometimes USB drives get mis-identified as "System" drives. To prevent you accidentally overwriting a system drive, we hide them from the drive-selection window unless "unsafe mode" is enabled.

> Yes, my Kodak card reader is specified for SD with no mention of SDHC. Looks like I'll have to buy a more modern one.

Ahh, good call. Luckily SD card readers are now **much** cheaper than they were 10 years ago :-)

GitUser275, 2017-12-06:
My SDHC compatible card reader arrived yesterday.   My first attempt to flash the image was with the card reader connected to an external USB hub.  That attempt took about 2 hours and failed validation.  Second attempt was with reader connected directly to a computer USB port.  That attempt took about 10 minutes and completed ok. 

Not enough data to reach any conclusion yet but I least I have another SDHC compatible port available.

lurch, 2017-12-06:
I guess the _huge_ speed reduction might be explained by the USB hub only being USB1.0 / USB1.1 whereas your computer's USB ports are USB2.0 (or USB3.0)?

Thanks for the feedback, good luck with the rest of your testing, and let us know if this issue can be closed.

lurch, 2018-02-28:
Closing this issue for now, but feel free to leave further comments if you're still having problems @GitUser275

closed 2018-02-28 by lurch