balena-io/etcher #1878: Etcher should not use internet when analytics disabled

ValdikSS, 2017-11-25:
- **Etcher version:** 1.1.2
- **Operating system and architecture:** Linux x86_64
- **Image flashed:** None

Etcher should not use internet when the user don't want it to. Currently there's no way to disable update check, for example.
Personally I don't see any point why image writing application should use internet at all. This is nonsense.

closed 2018-10-15 by jviotti

Hasshu, 2018-11-13:
It's v1.4.6 and Etcher still phones home like there's no tomorrow (on Windows, at least). How come this issue is closed?

LucianBuzzo, 2018-11-13:
@Hasshu I believe this issue has been closed in favor of https://github.com/balena-io/etcher/issues/2497

saitam1, 2019-07-19:
GDPR penalties so far http://www.enforcementtracker.com/