balena-io/etcher #1880: Invalid Version: static at new SemVer

LanceHaverkamp, 2017-11-27:
- **Etcher version:**
v1.1.2 - 2017-08-07
- **Operating system and architecture:**
Xubuntu 16.04 LTS
- **Image flashed:**
N/A error reported before image selection
- **Do you see any meaningful error information in the DevTools?**
Nothing happed pressing `Ctrl+Alt+I`

Error message at launch:

TypeError: Invalid Version: static
    at new SemVer (/tmp/.mount_7lKIUI/usr/bin/resources/app.asar/node_modules/etcher-latest-version/node_modules/semver/semver.js:293:11)
    at compare (/tmp/.mount_7lKIUI/usr/bin/resources/app.asar/node_modules/etcher-latest-version/node_modules/semver/semver.js:566:10)
    at Function.gt (/tmp/.mount_7lKIUI/usr/bin/resources/app.asar/node_modules/etcher-latest-version/node_modules/semver/semver.js:595:10)
    at bucketEntries.reduce.latest (/tmp/.mount_7lKIUI/usr/bin/resources/app.asar/node_modules/etcher-latest-version/index.js:93:20)
    at Array.reduce (native)
    at /tmp/.mount_7lKIUI/usr/bin/resources/app.asar/node_modules/etcher-latest-version/index.js:87:43
    at Parser.<anonymous> (/tmp/.mount_7lKIUI/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:484:18)
    at emitOne (events.js:96:13)
    at Parser.emit (events.js:188:7)
    at Object.onclosetag (/tmp/.mount_7lKIUI/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:445:26)
    at emit (/tmp/.mount_7lKIUI/usr/bin/resources/app.asar/node_modules/sax/lib/sax.js:640:35)
    at emitNode (/tmp/.mount_7lKIUI/usr/bin/resources/app.asar/node_modules/sax/lib/sax.js:645:5)
    at closeTag (/tmp/.mount_7lKIUI/usr/bin/resources/app.asar/node_modules/sax/lib/sax.js:905:7)
    at Object.write (/tmp/.mount_7lKIUI/usr/bin/resources/app.asar/node_modules/sax/lib/sax.js:1449:13)
    at Parser.exports.Parser.Parser.parseString (/tmp/.mount_7lKIUI/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:503:31)
    at Parser.parseString (/tmp/.mount_7lKIUI/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:7:59)
    at Object.exports.parseString (/tmp/.mount_7lKIUI/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:535:19)
    at /tmp/.mount_7lKIUI/usr/bin/resources/app.asar/node_modules/etcher-latest-version/index.js:80:9
    at $http.get.then (/tmp/.mount_7lKIUI/usr/bin/resources/app.asar/lib/gui/components/update-notifier/services/update-notifier.js:45:18)
    at processQueue (/tmp/.mount_7lKIUI/usr/bin/resources/app.asar/node_modules/angular/angular.js:16170:28)
    at /tmp/.mount_7lKIUI/usr/bin/resources/app.asar/node_modules/angular/angular.js:16186:27
    at Scope.$eval (/tmp/.mount_7lKIUI/usr/bin/resources/app.asar/node_modules/angular/angular.js:17444:28)

lurch, 2017-11-28:
@jhermsmeier Is it possible that this error occurred because the user was trying to run Etcher at the same time that you were uploading the builds for the latest version?

> `Ctrl+Alt+i` does open Dev Tools, but it means nothing to me

@LanceHaverkamp If you can still reproduce this error, please switch to the "Console" tab of the DevTools window, and then copy'n'paste here all the output you see.

jhermsmeier, 2017-11-28:
@LanceHaverkamp Are you running a snapshot build (i.e. from etcher-devel)? We can't see any evidence of that happening on the production version of v1.1.2 so far.

LanceHaverkamp, 2017-11-28:
tracker.js:10  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@resin.io
tracker.js:10 Tue Nov 28 2017 16:31:41 GMT-0700 (MST) Application start
tracker.js:10 Tue Nov 28 2017 16:31:41 GMT-0700 (MST) Checking for updates
https://usage.trackjs.com/usage.gif?token=032448bc3d9e4cffb1e9b43d29d6c142&…7a-95dcd9150878&application=etcher&x=25aeaea9-83e0-4c38-9c0f-d2bda6bb2973& Failed to load resource: net::ERR_INSECURE_RESPONSE
tracker.js:10 TypeError: Invalid Version: static
    at new SemVer (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/etcher-latest-version/node_modules/semver/semver.js:293:11)
    at compare (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/etcher-latest-version/node_modules/semver/semver.js:566:10)
    at Function.gt (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/etcher-latest-version/node_modules/semver/semver.js:595:10)
    at bucketEntries.reduce.latest (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/etcher-latest-version/index.js:93:20)
    at Array.reduce (native)
    at /tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/etcher-latest-version/index.js:87:43
    at Parser.<anonymous> (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:484:18)
    at emitOne (events.js:96:13)
    at Parser.emit (events.js:188:7)
    at Object.onclosetag (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:445:26)
    at emit (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/sax/lib/sax.js:640:35)
    at emitNode (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/sax/lib/sax.js:645:5)
    at closeTag (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/sax/lib/sax.js:905:7)
    at Object.write (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/sax/lib/sax.js:1449:13)
    at Parser.exports.Parser.Parser.parseString (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:503:31)
    at Parser.parseString (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:7:59)
    at Object.exports.parseString (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:535:19)
    at /tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/etcher-latest-version/index.js:80:9
    at $http.get.then (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/lib/gui/components/update-notifier/services/update-notifier.js:45:18)
    at processQueue (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/angular/angular.js:16170:28)
    at /tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/angular/angular.js:16186:27
    at Scope.$eval (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/angular/angular.js:17444:28)a.(anonymous function) @ tracker.js:10
tracker.js:10 TypeError: Invalid Version: static
    at new SemVer (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/etcher-latest-version/node_modules/semver/semver.js:293:11)
    at compare (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/etcher-latest-version/node_modules/semver/semver.js:566:10)
    at Function.gt (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/etcher-latest-version/node_modules/semver/semver.js:595:10)
    at bucketEntries.reduce.latest (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/etcher-latest-version/index.js:93:20)
    at Array.reduce (native)
    at /tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/etcher-latest-version/index.js:87:43
    at Parser.<anonymous> (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:484:18)
    at emitOne (events.js:96:13)
    at Parser.emit (events.js:188:7)
    at Object.onclosetag (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:445:26)
    at emit (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/sax/lib/sax.js:640:35)
    at emitNode (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/sax/lib/sax.js:645:5)
    at closeTag (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/sax/lib/sax.js:905:7)
    at Object.write (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/sax/lib/sax.js:1449:13)
    at Parser.exports.Parser.Parser.parseString (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:503:31)
    at Parser.parseString (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:7:59)
    at Object.exports.parseString (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:535:19)
    at /tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/etcher-latest-version/index.js:80:9
    at $http.get.then (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/lib/gui/components/update-notifier/services/update-notifier.js:45:18)
    at processQueue (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/angular/angular.js:16170:28)
    at /tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/angular/angular.js:16186:27
    at Scope.$eval (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/angular/angular.js:17444:28) undefineda.(anonymous function) @ tracker.js:10
tracker.js:10 TypeError: Invalid Version: static
[... cut ...]
    at $http.get.then (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/lib/gui/components/update-notifier/services/update-notifier.js:45:18)
    at processQueue (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/angular/angular.js:16170:28)
    at /tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/angular/angular.js:16186:27
    at Scope.$eval (/tmp/.mount_rWdTWc/usr/bin/resources/app.asar/node_modules/angular/angular.js:17444:28)a.(anonymous function) @ tracker.js:10
5https://capture.trackjs.com/capture?token=032448bc3d9e4cffb1e9b43d29d6c142 Failed to load resource: net::ERR_INSECURE_RESPONSE

LanceHaverkamp, 2017-11-28:
OK Problem solved:  Somehow I downloaded an older beta version.  I have no idea where that came from, but the _actual_ v1.1.2 appears to be working normally.
Sorry for the confusion!

closed 2017-11-28 by LanceHaverkamp