balena-io/etcher #1883: Fix OS window keyboard shortcuts

jhermsmeier, 2017-11-28:
Due to `electron.Menu.setApplicationMenu(null)`, native window shortcuts such as hiding or minimising the Window get removed.

lurch, 2017-11-28:
But Etcher is beautiful! Surely nobody would _ever_ want to hide it?

lurch, 2017-12-07:
In #1909 @joshlewis also requested:
⌘, - Open preferences (Command-comma)

Do the other platforms we support (Linux and Windows) have a similar keyboard shortcut for the same thing?

jhermsmeier, 2017-12-07:
> Do the other platforms we support (Linux and Windows) have a similar keyboard shortcut for the same thing?

Not that I'm aware of, but will double-check

lurch, 2017-12-11:
It occurred to me last night that `Ctrl` + `O` (open) might be a common keyboard shortcut that we could implement, to open the image-selection dialog?

lurch, 2017-12-11:
(sorry, fat-fingered the wrong button!)

jhermsmeier, 2018-01-05:
Closed in https://github.com/resin-io/etcher/pull/1888, released in v1.3.0

closed 2018-01-05 by jhermsmeier