balena-io/etcher #1889: stuck on Starting... after UAC dialog

terminatorbs, 2017-11-30:
1.2.0
win10 64bit
- n/a
-no

When starting Etch on a non- admin account, selecting an image, selecting the SD card and hitting Flash!:
It asks for admin privileges. I enter the admin privileges. The window asking for admin disappears, and Etch says "Starting..." forever. Nothing else happens. I have to X out of it, bringing some warning about damaging the SD Card.

Fix: Start etch as administrator ( right click. start as admin. enter admin user and password).

AhTn12, 2017-12-12:
I fixed this by right-clicking on etcher, and "Run As Administrator"

ema93a, 2017-12-21:
I had the same issue today, no matter how long I was waiting for it to advance. Funny thing is that this was during an assignment we didn't get to try because of this.

nathanle1406, 2017-12-23:
Had the same issue today, running as administrator fixed it.

wmhilton, 2017-12-23:
Thanks for sharing the "run as administrator" tip. You just saved my Raspberry Pi from oblivion. :)

jhermsmeier, 2018-01-18:
@jviotti this was fixed in https://github.com/resin-io/etcher/pull/1950, right?

jviotti, 2018-01-19:
@jhermsmeier It was fixed by simplifying the elevated process architecture and getting rid of the child writer proxy. The PR you're linking there was part of making that happen, but doesn't necessarily contain the fix.

jhermsmeier, 2018-01-19:
Sweet, I'll close this then!

closed 2018-01-19 by jhermsmeier

ShadowLNC, 2018-02-03:
#1950 was merged on Jan 5, Etcher v1.3.1 was released Jan 24 (UTC+1030). Assuming that the aforementioned merge is a parent of the 1.3.1 release build, then this issue persists, can it please be reopened?

deya-eldeen, 2018-02-06:
for you information
it is stuck on my macbook air 2012 that have high seira
it's ok on the other macbook air 2015 with seira
i used the same flash on both cases.

AlexGoranov, 2018-09-15:
Same problem on my Ubuntu 18.04, using the latest version of Etcher at the moment, 1.4.4.x84_64

LongLiveCHIEF, 2019-02-09:
Stuck for me running as administrator. Win 10, etcher v 1.4.9.

Falkenjack, 2019-03-24:
I have the same issue on Win7 running Etcher 1.5.5, running as administrator does not help. Meanwhile, Rufus works.

olluz, 2019-07-08:
Same here. Win10 64bit Etcher 1.5.51
Thanks for the Hint: With Rufus it works perfectly!

ekardon, 2020-04-25:
i had the same problem. i had a flash drive plugged in my system. after i safely remove sd-card and flash drive. i replugged the sd-card and it worked. about rufus, it doesn't even recognizes my sd-card. hope this helps

Litreily, 2020-11-01:
stuck starting on win10,  etcher 1.5.109.