balena-io/etcher #1890: Missing etcher-CLI examples

eranj, 2017-11-30:
CLI examples should be added to:
https://github.com/resin-io/etcher/blob/master/docs/CLI.md

lurch, 2017-12-01:
Is this just a generic request for documentation, or is there also some part of the Etcher CLI that you need help with?

eranj, 2017-12-01:
It's not just for me.
I mean examples like in the following link:
https://www.bountysource.com/issues/45325226-etcher-cli-help-output-is-misleading
How to set the source file and destination drive, with or without verification, unmount or not when flashing finished.

lurch, 2017-12-02:
Haha, that link you posted there is actually just a copy of #1452 ;-)

We've been focusing most of our efforts on the Etcher GUI so far, as that has **much** greater usage, but we'll try to add some polish (and documentation) to the Etcher CLI at some point soon.

closed 2018-10-15 by jviotti