balena-io/etcher #1893: Please also provide unzipped AppImages for download

probonopd, 2017-12-02:
This has been discussed before but has not been satisfactorily resolved. Currently the Etcher AppImages are only provided for download in zipped form, which 
* Cripples functionality of the optional `appimaged` daemon which does automatic desktop integration
* Prevents them from being included in the AppImageHub central directory of available AppImages, and hence from appearing in app stores and software centers like the NX Software Center
* Prevents them from being updated with binary delta updates using AppImageUpdate
Please also provide unzipped AppImages for download

Please also )additionally) provide the unzipped AppImages for download on GitHub Releases. Thanks!

lurch, 2017-12-02:
See #1842

probonopd, 2017-12-02:
@lurch 

> See #1842

I am aware of that and hence am suggesting here to provide an unzipped version __additionally__ on GitHub Releases. New users will download from the Etcher homepage anyway.

lurch, 2017-12-02:
Ah cool - I thought you were suggesting that https://etcher.io should offer both the zipped and unzipped AppImages, which would make things far too confusing for our users!

If all you're asking for is the unzipped AppImages to be provided on https://github.com/resin-io/etcher/releases along with the zipped AppImages, then I _think_ that should be okay.

lurch, 2017-12-02:
@jhermsmeier Thinking about this, if we do add the unzipped AppImages to GitHub Releases, we should also add them to S3 too.

lurch, 2017-12-09:
Unzipped AppImages are now included on https://github.com/resin-io/etcher/releases/tag/v1.2.1 :smiley:

closed 2017-12-09 by lurch

probonopd, 2017-12-09:
Thank you very much @lurch!

lurch, 2017-12-09:
Thanks due to @jhermsmeier too :-)