balena-io/etcher #1894: Document how to create images with metadata

probonopd, 2017-12-02:
Please document how to create images with metadata. I have an `.img` that I would like to distribute to Etcher users using the Etcher metadata features, but I cannot find any information about them.

Also, it is not clear how the `.img` file inside e.g., a `.tar.bz2` needs to be named in order for Etcher to recognize it. 

A Wiki page or documentation file for how to create Etcher-compliant archives of disk images is missing.

lurch, 2018-02-28:
We're no longer "officially supporting / promoting" metadata inside archives, are we @jhermsmeier ?

> Also, it is not clear how the .img file inside e.g., a .tar.bz2 needs to be named in order for Etcher to recognize it.

IIRC you can't have an image inside a `.tar.bz2` - what you'd want to do is have a `.img.bz2` file.

jhermsmeier, 2018-02-28:
Yup, that's correct @lurch – we've never had support for tarballs so far (but zip archives though), and the metadata + archive thing is deprecated.

jhermsmeier, 2018-02-28:
I'm going to close this for now, but if you feel like making those archives with metadata @probonopd, you can check out [ev3dev's releases](https://github.com/ev3dev/ev3dev/releases), which have a manifest & custom icon etc. There is a chance that support for this is going away in v2.0.0 though.

closed 2018-02-28 by jhermsmeier