balena-io/etcher #1895: .tar.bz2 image files

bearhoney, 2017-12-02:
Etcher seems to directly support ZIP files, but **no .tar.bz2 files** (as used e.g. by the NextcloudPi project).
Any volunteer to add this?

- **Etcher version: 1.2.0**
- **Operating system and architecture: Linux (Manjaro 17.1)**
- **Image flashed:  NextCloudPi_11-23-17.tar.bz2**
- **Do you see any meaningful error information in the DevTools? n/a**

jhermsmeier, 2017-12-02:
Just as a reference, the image can be obtained from https://ownyourbits.com/downloads/NextCloudPi_11-23-17/

lurch, 2017-12-02:
Hmmm, that `.tar.bz2` file only contains a single `NextCloudPi_11-23-17.img` file. I wonder why they didn't just create a simple `NextCloudPi_11-23-17.img.bz2` file?

lurch, 2017-12-02:
Ping @nacho who appears to be the https://github.com/nextcloud/nextcloudpi maintainer...
I'm not saying that Etcher won't support this feature, I'm just curious why you chose to create `NextCloudPi_11-23-17.tar.bz2` instead of the (more standard) `NextCloudPi_11-23-17.img.bz2` ?

jlabs, 2017-12-06:
I've just tried to flash a .img.bz2 file without success - was getting "ETIMEDOUT: connection timed out write" at around 80-90%.
Extracting the .img file inside of the .bz2 and using that worked fine.

I'm on 1.2.0 using High Sierra

lurch, 2017-12-07:
Hmmm, `.img.bz2` _should_ work. I'll investigate in the morning.
(BTW a `.tar.bz2` file is actually something significantly different from a `.img.bz2` file)

jhermsmeier, 2017-12-07:
@lurch @jlabs ETIMEDOUT probably has nothing to do with the file format, and reading or writing it, as it is specific to net sockets; This would hint at a write to the IPC channel timing out, I believe. Either way, this is not related to this issue, let's continue the discussion at https://github.com/resin-io/etcher/issues/1908

closed 2018-10-15 by jviotti

bearhoney, 2018-10-15:
Sorry, is there any closing comment? Is etcher now able to directly 
write  .tar.gz  (or .tgz) compressed files? Or is the thread being 
closed as feature not {feasible / important / desired}?

Thanks,
Bernhard

lurch, 2018-10-16:
@bearhoney You might want to look at #2338 ?