balena-io/etcher #1898: .imz file support

zambelly, 2017-12-03:
Etcher doesn't support .imz files created by [USB Image Tool](http://www.alexpage.de/usb-image-tool/). It's just a zip file, and you can rename it to .zip and it works, but it would be nice if you didn't have to do that.
- **Etcher version: 1.2.0**
- **Operating system and architecture: Windows 10 Pro (x64)**
- **Image flashed: ubuntu.imz**
- **Do you see any meaningful error information in the DevTools? n/a**

lurch, 2017-12-04:
With no disrespect to the author of USB Image Tool, I'd rather see #266 implemented than add support for whatever weird and wonderful image extensions 3rd party tools happen to create.
But that's just my personal opinion, and the other Etcher developers may disagree ;-)

jhermsmeier, 2017-12-04:
@lurch I can't see how that has anything to do with this issue?

lurch, 2017-12-04:
Sorry, perhaps I should have given a bit more context. I clicked on the USB Image Tool above, and it's a GUI program for reading and writing disk images from and to USB drives. A bit like what Etcher does, without the 'reading' part...

closed 2018-10-15 by jviotti