balena-io/etcher #1899: Flash Drive Dead

thor314, 2017-12-04:
*Forewarning: Am a beginner, know very little.
I was formatting a pair of USBs for Ubuntu and Lubuntu with Etcher.
I followed the same procedure with both USBs.  The Ubuntu one went correctly.  The Lubuntu one failed, and I don't know how or how to reset my USB.

I tried to reset the USB with this guide on [Tails](https://tails.boum.org/doc/first_steps/reset/mac/index.en.html)
For `sudo fdisk -iy /dev/disk1s2`, Terminal returned
fdisk: could not open MBR file /usr/standalone/i386/boot0: No such file or directory
diskutil list returns:
```
/dev/disk1 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *16.0 GB    disk1
   1:                 Apple_Boot                         8.4 MB     disk1s1
   2:                  Apple_HFS                         16.0 GB    disk1s2
```

zambelly, 2017-12-04:
If you want to reformat the drive, this command will reformat disk1 as FAT32 and give it the name LUBUNTU:
`diskutil eraseDisk FAT32 LUBUNTU /dev/disk1`
After reformatting, you can try flashing again with Etcher
Make sure `/dev/disk1` is the drive you actually want to reformat, because you can accidentally reformat something else.

lurch, 2017-12-04:
This may help too https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives