balena-io/etcher #1901: "access denied" right afater installation

dwfunk4475, 2017-12-04:
Etcher-Setup-1.2.0-x64.exe
Win  7 Ent.

This error has shown up on installation of Etcher onto my system.
Fresh download of app from website, installed it and this error pops up.
The path displayed in the error is not my system.  I have no idea where that path came from.  I've only installed this package and it errors without even running it.

[image]

lurch, 2017-12-05:
Are you behind some sort of network proxy? If you click "OK" does Etcher continue to run, or does it close?

lurch, 2017-12-05:
A similar `EPROTO` error was spotted in #1808

dwfunk4475, 2017-12-05:
Yes it continues to run and flashes ok.

Why would proxy matter?

-david

lurch, 2017-12-06:
Etcher checks for updates at startup (using an HTTPS / SSL connection to Amazon S3) and I wondered if you might have a proxy that is blocking or restricting HTTPS, which might explain the openssl error?

If we can figure out a way for us to reproduce the error ourselves, then that makes it much easier for us to fix it (or workaround it) :-)

dwfunk4475, 2017-12-06:
Interesting.  No proxy, not even on a network.

As another data point that might be helpful, I installed on a laptop:

Etcher-Setup-1.2.0-x64.exe
Win 10

and get this error when I start the program:

[image: Inline image 1]

I can click "OK" and the error dialog goes away and Etcher will image an SD
card.

This laptop is not connected to any network.

-david

lurch, 2017-12-07:
@dwfunk4475 Your image didn't appear - I don't think GitHub can cope with images in emails, you need to go to the GitHub webinterface itself to add images.

dwfunk4475, 2017-12-07:
attaching image  . . .
[image]

closed 2017-12-07 by dwfunk4475

lurch, 2017-12-07:
Wow, that's a pretty bizarre error message! Is it possible your Etcher download or installation has become corrupted somehow?

dwfunk4475, 2017-12-07:
That's something I can check. I'll re-download the file from a different
location and try again. I've used the same downloaded file on both machines.

I can clear the dialog boxes on either error and Etcher seems to run just
fine. I have written a couple of SD cards and they have turned out OK.

-david