balena-io/etcher #1907: Freezes Windows 10 PC on Start

Abyss2381, 2017-12-06:
- **Etcher version:1.20**
- **Operating system and architecture: Windows 10 Pro 1607 14393.1914 64bit**
- **Image flashed: Recalbox 17.12.02**
- **Do you see any meaningful error information in the DevTools? No, the pc freezes completely and I have to reboot.**

I start Etcher, choose the image, and then choose the SD card. Once I click Flash, the button changes to orange and it's label changes to "Starting". After 10 seconds, my pc becomes completely unresponsive and I have to reboot. I have tried to re-download Etcher, downloaded a previous version, changed to a Retropie Image, and changed out SD cards, as well as interface device and usb port. Always the same. 10 seconds in, the pc freezes completely and requires a hard power off to restart. Tried Etcher in a virtual machine running Ubuntu 17.10  on the same pc, same thing. It completely froze the system, requiring a reboot. I do not have access to another pc to test if it's something in my hardware, which I am sure it is. Also, I did try loading it in Windows without anything else loaded (I.E. Safe startup, and ran bare minimum environment), same result.

Specs:

Intel 4770K, not overclocked.
Fatal1ty Z87 Killer Bios V. 2.10
16GB (2x8gb matched pair) Patriot DDR3 - 1600  10/10/10/27
OS Drive: Crucial MX300 2.5 SSD CT525MX300SSD1 525GB
Storage: Intel Raid 5 (3x Seagate 3TB IronWolf NAS SATA 6Gbs NCQ w/64MB Cache)
Video Card: Asus Strix OC Nvidia GTX 1080ti Driver V. 388.31

I'd get a DevTools readout if I could, but as it freezes my entire PC, and I have to power off, I'm not finding anything or any logs.

Abyss2381, 2017-12-06:
Oh, I forgot:
SD cards used are Sandisk Ultra 64GB, 32GB, and tried a Silicon Power SP 32GB UHS-1 card as well.

lurch, 2017-12-07:
Can you paste the DevTools log from immediately _before_ you click the Flash button, in case that allows us to spot anything suspicious?

@jhermsmeier @jviotti On a side-note, have we ever tested how `drivelist` works with a RAID setup?

cebess, 2017-12-09:
Same problem here. Windows 10 (latest release), tried multiple SD cards. Win32diskimager was able to apply the image, so it doesn't seem like an issue with the SD card...

Abyss2381, 2017-12-11:
It was the later I do believe (RAID setup). I was replacing my RAID with an external NAS, and was using a Linux Live DVD to move all the stuff around. Decided to try Etcher while doing so. It worked flawlessly. I further went in and removed my raid setup after removing the array and reverted my bios to AHCI instead of RAID. It works in Windows 10 now. I'll copy paste the DevTools log later as I still have a RAID Pc at work not functioning.

kpalf, 2017-12-23:
Same issues here.  I'm on Win10_Home64 1709 16299.125, running as a non-admin user.  This is an HP Spectre x360 laptop, and the SD card is a Transcend 16GB.  Image was 2017-11-29-raspbian-stretch.zip 

Running Etcher (1.2.1) as non-admin user when I choose the image and then select flash I get asked to elevate, but no progress other than the button changing to orange and "starting...".  I've also had the complete OS lockup described above.  

Repeating the same thing but starting Etcher process as Admin, and it seems to work.

lurch, 2018-02-28:
Could you please re-test with Etcher 1.3.1 and let us know if that works any differently / better? :)

efidoman, 2018-03-03:
I'm running 1.3.1 and still experiencing the same issue

bac621, 2018-03-04:
same issue

closed 2018-10-15 by jviotti

CallMeFoxie, 2019-10-05:
FYI still happening with 1.5.57 on my system - sadly I found this thread after rolling back BIOS, Windows Update and drivers and doing a refresh setup of the PC.

Computer:
- Ryzen 2700
- MSI X570A Pro
- 32GB RAM
- Win10 x64 (1903, latest updates)
- 3x SSD
- 3x HDD **from which two are in RAID1**

tested several SD cards and 2 readers (on USB3 and USB2)

Hope that somebody with the same problem finds this thread before going crazy with their computer.

Xstoudi, 2020-04-11:
Same problem.

keen9, 2020-07-02:
Oh man oh man, this issue... Created this account just to post the solution that worked for me. My computer worked flawlessly as long as I had w7, and then when I uppgraded to w10, these strange "my computer freezes"-problems started to occur (hard reset only afterwards). Flashing with Etcher, installing printer driver (Samsung ml-2165), installing Logitech G-hub, they all ended up in complete freezes. 

Then I stumbeld over this: [https://forum-en.msi.com/index.php?topic=324209.0](https://forum-en.msi.com/index.php?topic=324209.0). The solution was to disconnect my dvd-reader from the motherboard (it should be ok to connect it to another sata-port, but I didn't need it anyway). Oh lo and behold, suddenly all my issues were gone! No more freezes, and everything that previously caused a freeze worked like a charm. 

Hope this will come to use for anyone!

thundron, 2020-07-07:
Thanks for sharing your solution @keen9. While others may have different hardware, the concept is the same: there's something with a faulty driver or connection which is interfering with other resources