balena-io/etcher #1908: ETIMEDOUT when flashing compressed image

jhermsmeier, 2017-12-07:
- **Etcher version:** 1.2.0
- **Operating system and architecture:** Mac OS High Sierra
- **Image flashed:** unknown.img.bz2

Quoting @jlabs from https://github.com/resin-io/etcher/issues/1895#issuecomment-349764194:

> I've just tried to flash a .img.bz2 file without success - was getting "ETIMEDOUT: connection timed out write" at around 80-90%.
Extracting the .img file inside of the .bz2 and using that worked fine.
> 
> I'm on 1.2.0 using High Sierra

jhermsmeier, 2017-12-07:
@jlabs is the image you've been flashing publicly available? If yes, could you link us to it, so we can try to reproduce / investigate this?

lurch, 2017-12-07:
@jhermsmeier In gitter, @jlabs mentioned he was trying to flash a "HASS.io image". So I've downloaded the https://github.com/home-assistant/hassio-build/releases/download/1.1/resinos-hassio-1.1-raspberrypi3.img.bz2 image and have been testing with that. This is a 1.7GB image file which bzips down to just 136MB (so it obviously contains a fair amount of empty space).
Even on my monster desktop PC (Intel i7-5820K CPU @ 3.30GHz with 16GB ram) I see some "unusual" behaviour. Relevant snippets from my DevTools log (with the raven.js removed for clarity):
```
Thu Dec 07 2017 11:41:05 GMT+0000 (GMT) Progress (write): 73% at 5.62 MB/s (eta 19.642797496220837s)
Thu Dec 07 2017 11:41:05 GMT+0000 (GMT) Progress (write): 73% at 5.61 MB/s (eta 21.73930102632246s)
Thu Dec 07 2017 11:41:06 GMT+0000 (GMT) Progress (write): 74% at 5.65 MB/s (eta 21.456189414327458s)
Thu Dec 07 2017 11:41:07 GMT+0000 (GMT) Progress (write): 75% at 5.88 MB/s (eta 19.54300319447237s)
Thu Dec 07 2017 11:41:27 GMT+0000 (GMT) Progress (write): 75% at 5.7 MB/s (eta 20.448437476342963s)
Thu Dec 07 2017 11:41:27 GMT+0000 (GMT) Progress (write): 75% at 5.15 MB/s (eta 20.05991171334536s)
Thu Dec 07 2017 11:41:28 GMT+0000 (GMT) Progress (write): 75% at 5.34 MB/s (eta 21.022787475585936s)
Thu Dec 07 2017 11:41:35 GMT+0000 (GMT) Progress (write): 75% at 5.49 MB/s (eta 24.93188985188802s)
Thu Dec 07 2017 11:41:35 GMT+0000 (GMT) Progress (write): 76% at 5.47 MB/s (eta 22.329116821289062s)
Thu Dec 07 2017 11:41:36 GMT+0000 (GMT) Progress (write): 77% at 5.83 MB/s (eta 21.37259508215863s)
```
Notice how everything is chugging along quite happily (with updates at least once a second), but then when it gets to 75% there's a huge pause of 20 seconds between updates. During this time, the Etcher GUI appears to 'hang' and a user might think that it's crashed? And it's quite easy to imagine that on slower CPUs (that would take even longer to decompress the bzip2 data), it's indeed possible that this delay between updates is enough to trigger a `ETIMEDOUT` from the IPC connection :-/

And also:
```
Thu Dec 07 2017 11:42:09 GMT+0000 (GMT) Progress (write): 99% at 8.66 MB/s (eta 0.533611692231277s)
Thu Dec 07 2017 11:42:09 GMT+0000 (GMT) Progress (write): 99% at 9.87 MB/s (eta 0.16190950260605924s)
Thu Dec 07 2017 11:42:10 GMT+0000 (GMT) Progress (write): 100% at 8.6 MB/s (eta 0s)
Thu Dec 07 2017 11:43:18 GMT+0000 (GMT) Progress (check): 4% at 87.29 MB/s (eta 19.66066066066066s)
Thu Dec 07 2017 11:43:18 GMT+0000 (GMT) Progress (check): 11% at 138.02 MB/s (eta 11.567426400759734s)
Thu Dec 07 2017 11:43:19 GMT+0000 (GMT) Progress (check): 18% at 166.2 MB/s (eta 8.85173501577287s)
```
Etcher gets to the end of reading through the compressed data, and so outputs "100% write progress" which means that the Etcher GUI displays the "Finishing..." button. However the uncompressed data is still being written to disk, and this means that the "Finishing..." button is displayed for a whole 68 seconds (!) before the button changes from "Finishing..." to "Validating". An impatient user might again assume that Etcher has crashed at this point.

I believe both these "big pauses" are explained by #1171 (with a more in-depth explanations [here](https://github.com/resin-io/etcher/issues/638#issuecomment-285483310) and [here](https://github.com/resin-io/etcher/pull/1587#issuecomment-314773901) ) which I suggested a workaround for [here](https://github.com/resin-io/etcher/pull/1863#issuecomment-347513569).

lurch, 2017-12-07:
Perhaps this relates (indirectly?) to #1691 too?

jviotti, 2017-12-08:
I ran a small test where I'd open start the IPC server, connect to it, but don't send any messages for up to 20 minutes, but never got a timeout. Also, from Node.js documentation:

> By default net.Socket do not have a timeout.

And `node-ipc` is not setting one either.

jhermsmeier, 2017-12-08:
@jviotti it's the write() that times out, not the socket / connection itself from what I understand

wdool, 2018-01-02:
I had the same experience... Stupid but simple solution.. 
Instead of using my built in SD Card reader (Imac) I used my USB Card reader.

pizzapants, 2018-03-03:
Has this actually been fixed? Just had this same error trying to flash the compressed hass.io image with the latest etcher on my macbook

thebaron, 2018-03-04:
As another data point, Etcher 1.3.1 hangs at 74%, then again at ~85% on the latest 1.2 version of the bz2-compressed hassio image released earlier today. I was able to get it to complete flashing by using a USB reader instead of the built-in SD slot on my iMac. It hung a long time at Finalizing… as well, and has just now begun validating. Validation, unlike writing, doesn't seem to get "stuck" along the way, it's been a steady flow. 

As another data point, which may or may not help isolate the issue, I used bunzip2 on the CLI to bunzip2 -c image.img.bz2 > image.img and then used that uncompressed image thru Etcher 1.3.1 to the internal SD reader and, while seemingly slower (?) at flashing than the compressed image was (the data rate is the same of ~8MB/s) it went thru the flashing process fine. It validated much more quickly (~50MB/s) likely owing to the far superior SD slot than my USB adapter. But there were no delays using the uncompressed image.

jhermsmeier, 2018-03-04:
@thebaron those delays, or "getting stuck" when flashing compressed images happens due to the image compressing a very large region into a very small chunk (mostly zero-filled regions of the image), and because we can't display the write-progress for some compression schemes (because they have no reliable indication of their uncompressed size), the progress for them is based on how much of the source has been read - meaning it's normal for it to appear "stuck" in those areas, and is not really a bug, unless it stays stuck.

thebaron, 2018-03-04:
Hey @jhermsmeier thanks for the reply! That makes 100% sense to me; I was hoping it would turn out to be more helpful but that does make perfect sense. It's interesting that the in-Mac reader seems to have problems when the USB-stick didn't. In any event, the uncompressed image worked either way, so it's easy enough to work around it's just ... weird.

jhermsmeier, 2018-03-05:
@pizzapants No, the ETIMEDOUT hasn't been fixed yet – we only recently got a lead on what could be causing it in https://github.com/resin-io/etcher/issues/1793#issuecomment-369329248 (we couldn't reproduce it with any of our machines so far, which makes fixing it a bit tricky)

jhermsmeier, 2018-03-05:
@thebaron yeah, we'll be looking into making it less awkward, and more consistent in the future, but given what I mentioned, it might not work out for everything unfortunately.

PTerblanche, 2018-03-13:
I am having the same problem. Trying to write image to SD card with Etcher but get a red error code at 84% and then etcher stops. How can I get the image file to the sd card now ?

lurch, 2018-03-13:
@PTerblanche The simple workaround for now is to manually uncompress the image first, and then use Etcher to write the uncompressed image to your SD card.

PTerblanche, 2018-03-13:
Hi

How do i do that ?

thebaron, 2018-03-14:
Hey, @PTerblanche I just launched a terminal window used the bunzip2 command for .bz2 images (or the gunzip command for .gz compressed images):

`bunzip2 -c image.img.bz2 > image.img`
`gunzip -c image.img.gz >  image.img`

Then drop the .img onto Etcher and you should be good to go!

PTerblanche, 2018-03-14:
Hi there

I get this result

Pieters-MacBook-Pro:downloads pieterterblanche$ bunzip2 -c resinos-hassio-1.1-raspberrypi3.img.bz2 > image.img gunzip -c image.img.gz > hassio.img
bunzip2: Can't open input file gunzip: No such file or directory.
bunzip2: Can't open input file image.img.gz: No such file or directory.
Pieters-MacBook-Pro:downloads pieterterblanche$

optimusxcrime, 2018-03-14:
Extract the file

xz -d odroidimagegoeshere.img.xz

Write the image file to the SD card

sudo dd if=odroidimagegoeshere.img.xz of=/dev/sdc bs=4M

optimusxcrime, 2018-03-14:
> ons/unsubscribe-auth/AiO0hUWh1oVqcQ3EHQ4w1lZB0NBb6ecnks5teIpmgaJpZM4Q5SKW

make sure you put the right number down that represents your removable
storage.

/dev/sdc"righthere"

Or you could wipe your hdd...

Cheers!

PTerblanche, 2018-03-14:
I will try this thx.

> odroidimagegoeshere.img.xz will be the hassio img.bz2 file name correct?

optimusxcrime, 2018-03-14:
shouldn't be any spaces. and make sure you put the entire path to the img.
/home/odroid/image.img.xz
or terminal will return with error. unable to find file or directory.

optimusxcrime, 2018-03-14:
didnt know it wasnt an xz file. use gunzip to exract it

optimusxcrime, 2018-03-14:
http://odroid.us/mediawiki/index.php?title=Step-by-step_Ubuntu_SD_Card_Setup

this is a fool proof guide

lurch, 2018-03-14:
@PTerblanche In his comment, @thebaron was giving you two alternative commands; he was basically saying "if you have a .bz2 file then run `bunzip2 -c image.img.bz2 > image.img` but if you have a .gz file then run `gunzip -c image.img.gz > image.img` "
But it looks like you copied both the commands and tried to run them both at once, which won't work! :wink: 

The command you'd want to use is
```
bunzip2 -c resinos-hassio-1.1-raspberrypi3.img.bz2 > resinos-hassio-1.1-raspberrypi3.img
```
and then use Etcher to write the .img file to your SD card.

thebaron, 2018-03-15:
Just do this first one:

`bunzip2 -c resinos-hassio-1.1-raspberrypi3.img.bz2 > image.img`

closed 2018-10-15 by jviotti