balena-io/etcher #1909: Feature request: support for some extremely common macOS keyboard shortcuts

joshlewis, 2017-12-07:
There are a handful of universal commands on macOS that your app should support, to feel right to an experienced Mac user. I'd recommend supporting these in your app:

⌘-H  -  Hide the app  (Command-H)
⌥⌘-H  -  Hide all other apps (Option-Command-H)
⌘,  -  Open preferences (Command-comma)

Since the app could support those features, it would be nice if it did. They're standard on all Mac apps.  You'll notice that other 3rd party apps often support all three, e.g. Chrome.

lurch, 2017-12-07:
Is this related to #1888 @jhermsmeier ?

jhermsmeier, 2017-12-07:
Yes, this will be fixed by #1888 
Duplicate of #1883

jhermsmeier, 2017-12-07:
Duplicate of #1883

closed 2017-12-07 by jhermsmeier

joshlewis, 2017-12-07:
I don't see any mention of the ⌘,  preferences shortcut on any of those. Maybe this issue could be preserved to track that feature?

lurch, 2017-12-07:
@joshlewis I've added an extra comment to #1883 ... :-)

joshlewis, 2017-12-07:
Awesome! Thank you. 🙇

jhermsmeier, 2017-12-07:
Oh sorry, I missed that – totally should add that as well, thanks @joshlewis!

jhermsmeier, 2018-01-05:
Fixed by https://github.com/resin-io/etcher/pull/1888, released in v1.3.0