balena-io/etcher #1911: etcher doesn't auto unmount drives after successful flash on Ubuntu 17.10

shaunmulligan, 2017-12-07:
- **Etcher version:** v1.2.1 (but also happened on 1.2.0
- **Operating system and architecture:** Ubuntu 17.10 (Linux Blade-Stealth 4.13.0-17-generic #20-Ubuntu SMP Mon Nov 6 10:04:08 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux)
- **Image flashed:** resinOS v2.7.8 for the raspberry pi 3
- **Do you see any meaningful error information in the DevTools?** No errors that I can see.

lurch, 2017-12-08:
@jviotti I've been able to reproduce a similar problem using just https://github.com/resin-io-modules/mountutils and on Ubuntu 14.04 (i.e. this isn't isolated to Ubuntu 17.10).

1. Flash resinos to an SD card (in this particular example I'm using https://files.resin.io/resinos/raspberrypi3/2.3.0%2Brev1.dev/image/resin.img.zip )
2. Remove and reinsert the card, and all 4 partitions should auto-mount - if they don't, click on each of them in the Nautilus (file explorer) sidebar so that they're now mounted
3. Run a simple script like the one shown at https://github.com/resin-io-modules/mountutils#mountutilsunmountdiskdevice-callback
4. _Sometimes_ all 4 partitions will unmount, but most of the time only 3 of the partitions will unmount!

This can also happen when Etcher tries to unmount the drive when you click on the "Flash" button, which makes a couple of error dialogs pop up:
[image]

[image]

Example 'debug' output shown below...
```
$ cat /proc/mounts
sysfs /sys sysfs rw,nosuid,nodev,noexec,relatime 0 0
proc /proc proc rw,nosuid,nodev,noexec,relatime 0 0
udev /dev devtmpfs rw,nosuid,relatime,size=8106488k,nr_inodes=2026622,mode=755 0 0
devpts /dev/pts devpts rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000 0 0
tmpfs /run tmpfs rw,nosuid,noexec,relatime,size=1625444k,mode=755 0 0
/dev/md1 / ext4 rw,relatime,errors=remount-ro,stripe=256,data=ordered 0 0
securityfs /sys/kernel/security securityfs rw,nosuid,nodev,noexec,relatime 0 0
tmpfs /dev/shm tmpfs rw,nosuid,nodev 0 0
tmpfs /run/lock tmpfs rw,nosuid,nodev,noexec,relatime,size=5120k 0 0
tmpfs /sys/fs/cgroup tmpfs ro,nosuid,nodev,noexec,mode=755 0 0
cgroup /sys/fs/cgroup/systemd cgroup rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd 0 0
pstore /sys/fs/pstore pstore rw,nosuid,nodev,noexec,relatime 0 0
cgroup /sys/fs/cgroup/hugetlb cgroup rw,nosuid,nodev,noexec,relatime,hugetlb,release_agent=/run/cgmanager/agents/cgm-release-agent.hugetlb 0 0
cgroup /sys/fs/cgroup/cpuset cgroup rw,nosuid,nodev,noexec,relatime,cpuset,clone_children 0 0
cgroup /sys/fs/cgroup/pids cgroup rw,nosuid,nodev,noexec,relatime,pids,release_agent=/run/cgmanager/agents/cgm-release-agent.pids 0 0
cgroup /sys/fs/cgroup/perf_event cgroup rw,nosuid,nodev,noexec,relatime,perf_event,release_agent=/run/cgmanager/agents/cgm-release-agent.perf_event 0 0
cgroup /sys/fs/cgroup/memory cgroup rw,nosuid,nodev,noexec,relatime,memory 0 0
cgroup /sys/fs/cgroup/blkio cgroup rw,nosuid,nodev,noexec,relatime,blkio 0 0
cgroup /sys/fs/cgroup/cpu,cpuacct cgroup rw,nosuid,nodev,noexec,relatime,cpu,cpuacct 0 0
cgroup /sys/fs/cgroup/freezer cgroup rw,nosuid,nodev,noexec,relatime,freezer 0 0
cgroup /sys/fs/cgroup/net_cls,net_prio cgroup rw,nosuid,nodev,noexec,relatime,net_cls,net_prio 0 0
cgroup /sys/fs/cgroup/devices cgroup rw,nosuid,nodev,noexec,relatime,devices 0 0
systemd-1 /proc/sys/fs/binfmt_misc autofs rw,relatime,fd=33,pgrp=1,timeout=0,minproto=5,maxproto=5,direct 0 0
mqueue /dev/mqueue mqueue rw,relatime 0 0
hugetlbfs /dev/hugepages hugetlbfs rw,relatime 0 0
debugfs /sys/kernel/debug debugfs rw,relatime 0 0
fusectl /sys/fs/fuse/connections fusectl rw,relatime 0 0
binfmt_misc /proc/sys/fs/binfmt_misc binfmt_misc rw,relatime 0 0
cgmfs /run/cgmanager/fs tmpfs rw,relatime,size=100k,mode=755 0 0
tmpfs /run/user/1000 tmpfs rw,nosuid,nodev,relatime,size=1625444k,mode=700,uid=1000,gid=1000 0 0
gvfsd-fuse /run/user/1000/gvfs fuse.gvfsd-fuse rw,nosuid,nodev,relatime,user_id=1000,group_id=1000 0 0
/dev/md1 /var/lib/docker/aufs ext4 rw,relatime,errors=remount-ro,stripe=256,data=ordered 0 0
gvfsd-fuse /home/qpc/.gvfs fuse.gvfsd-fuse rw,nosuid,nodev,relatime,user_id=0,group_id=0 0 0
/dev/sdc1 /media/qpc/resin-boot vfat rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,showexec,utf8,flush,errors=remount-ro 0 0
/dev/sdc2 /media/qpc/resin-rootA ext4 rw,nosuid,nodev,relatime,data=ordered 0 0
/dev/sdc5 /media/qpc/resin-state ext4 rw,nosuid,nodev,relatime,data=ordered 0 0
/dev/sdc6 /media/qpc/resin-data ext4 rw,nosuid,nodev,relatime,data=ordered 0 0
$ sudo MOUNTUTILS_DEBUG=1 node test.js 
[mountutils] Reading /proc/mounts
[mountutils] Mount point found: sysfs
[mountutils] Mount point found: proc
[mountutils] Mount point found: udev
[mountutils] Mount point found: devpts
[mountutils] Mount point found: tmpfs
[mountutils] Mount point found: /dev/md1
[mountutils] Mount point found: securityfs
[mountutils] Mount point found: tmpfs
[mountutils] Mount point found: tmpfs
[mountutils] Mount point found: tmpfs
[mountutils] Mount point found: cgroup
[mountutils] Mount point found: pstore
[mountutils] Mount point found: cgroup
[mountutils] Mount point found: cgroup
[mountutils] Mount point found: cgroup
[mountutils] Mount point found: cgroup
[... 156 lines cut ...]
    mountpoints: [],
    raw: '/dev/sda',
    protected: false,
    system: true },
  { device: '/dev/sdb',
    displayName: '/dev/sdb',
    description: 'Samsung SSD 850',
    size: 256060514304,
    mountpoints: [],
    raw: '/dev/sdb',
    protected: false,
    system: true },
  { device: '/dev/sdc',
    displayName: '/dev/sdc',
    description: 'SD Transcend',
    size: 7969177600,
    mountpoints: 
     [ { path: '/media/qpc/resin-data' },
       { path: '/media/qpc/resin-boot' },
       { path: '/media/qpc/resin-rootA' },
       { path: '/media/qpc/resin-state' } ],
    raw: '/dev/sdc',
    protected: false,
    system: false } ]
```

lurch, 2017-12-08:
> Looking at https://github.com/resin-io-modules/mountutils/blob/master/src/linux/functions.cpp#L23 I can't see any obvious reason why /dev/sdc6 is getting skipped :-/

Oooh! It's suddenly occurred to me (flash of inspiration) that maybe it's invalid to try and call `umount2` while iterating over `/proc/mounts` with `getmntent` because I guess `umount2` will itself be causing modifications to `/proc/mounts`, which means that the file-pointer being used by `getmntent` ends up in an invalid location? (i.e. we've got a race condition, and trying to unmount 4 partitions inserts enough 'delay' to trigger it)
Is this is indeed the case, we should instead use `getmntent` to iterate over the entirety of `/proc/mounts` first, building up an array of mountpoints belonging to the disk that we're interested in, and then _afterwards_ `umount2` them all :-)

commit 1f838b4b (2018-01-23): upgrade(package): Bump mountutils 1.3.8 -> 1.3.10 (#1983)

This updates `mountutils` from 1.3.8 -> 1.3.10;

- fix(linux): Fix partial unmounts on Linux
- fix(windows): Link to appropriate libraries

Change-Type: patch
Changelog-Entry: Fix incomplete unmounts after flashing on Linux
Fixes #1911
closed 2018-01-23 by jhermsmeier