balena-io/etcher #1914: Categories= key missing in Linux desktop file

probonopd, 2017-12-09:
The `etcher.desktop` file should include a line

```
Categories=Utility;
```

References:
* https://travis-ci.org/AppImage/appimage.github.io/builds/313897555#L709
* https://github.com/AppImage/appimage.github.io/pull/199#partial-pull-merging

---

The desktop file is lacking a `Categories=` entry in the `*.desktop` file.

According to the [menu spec](https://standards.freedesktop.org/menu-spec/latest/apa.html),

> By including one of the Main Categories in an application's desktop entry file, the application will be ensured that it will show up in a section of the application menu dedicated to this category. If multiple Main Categories are included in a single desktop entry file, the entry may appear more than once in the menu. 

Hence, please add at least one of the following in the `Categories=` key.

Main Category | Description | Notes
-- | -- | --
AudioVideo | Application for presenting, creating, or processing multimedia (audio/video) |  
Audio | An audio application | Desktop entry must include AudioVideo as well
Video | A video application | Desktop entry must include AudioVideo as well
Development | An application for development |  
Education | Educational software |  
Game | A game |  
Graphics | Application for viewing, creating, or processing graphics |  
Network | Network application such as a web browser |  
Office | An office type application |  
Science | Scientific software |  
Settings | Settings applications | Entries may appear in a separate menu or as part of a 		"Control Center"
System | System application, "System Tools" such as say a log viewer or network monitor |  
Utility | Small utility application, "Accessories"

In addition, you could specify one or more from the longer list of [Additional Categories](https://standards.freedesktop.org/menu-spec/latest/apas02.html).

Please test the result with `desktop-file-validate` and make sure it passes.

__electron-builder offers native support for this since [v19.22.1](https://github.com/electron-userland/electron-builder/releases/tag/v19.22.1).__. Please see the `category` key at https://www.electron.build/configuration/linux-other.

jhermsmeier, 2017-12-11:
This will be fixed by https://github.com/resin-io/etcher/pull/1905 as our electron-builder config already sets the category to "Utility"

jhermsmeier, 2018-01-18:
Closing, as https://github.com/resin-io/etcher/pull/1905 has been merged, and this should be fixed with the next release.

closed 2018-01-18 by jhermsmeier