balena-io/etcher #1915: Couldn't scan the drives: An unknown error occurred

lurch, 2017-12-10:
- **Etcher version:** 1.2.1
- **Operating system and architecture:** Windows7 x64
- **Image flashed:** _any_
- **Do you see any meaningful error information in the DevTools?** Yes... ;-)
```
[snipped the usual stuff]
Sun Dec 10 2017 22:39:06 GMT+0000 (GMT Standard Time) Progress (check): 96% at 11.51 MB/s (eta 0.2619589977220957s)
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 Sun Dec 10 2017 22:39:06 GMT+0000 (GMT Standard Time) Progress (check): 100% at 11.02 MB/s (eta 0s)
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] \\.\PHYSICALDRIVE1
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Checking whether drive is ejectable: C
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] The drive is fixed
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Opening drive letter handle: C
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Closing drive letter handle
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Continuing with the next available letter
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Checking whether drive is ejectable: D
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] The drive is fixed
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Opening drive letter handle: D
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Closing drive letter handle
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Continuing with the next available letter
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Checking whether drive is ejectable: E
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] The drive is fixed
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Opening drive letter handle: E
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Closing drive letter handle
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Continuing with the next available letter
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Checking whether drive is ejectable: F
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] The drive is removable
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Opening drive letter handle: F
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Closing drive letter handle
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Drive letter device matches
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Ejecting drive letter
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Creating volume handle
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Volume is not mounted
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Retrying
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Ejecting drive letter
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Creating volume handle
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Volume is not mounted
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Retrying
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Ejecting drive letter
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Creating volume handle
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Volume is not mounted
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Retrying
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Continuing with the next available letter
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Checking whether drive is ejectable: Q
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] The drive is fixed
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Opening drive letter handle: Q
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Closing drive letter handle
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Continuing with the next available letter
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 [mountutils] Unmount complete
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 Sun Dec 10 2017 22:39:11 GMT+0000 (GMT Standard Time) Done ({"image":"C:\\Users\\Andrew\\Downloads\\alpine-standard-3.5.2-x86_64.iso","drive":{"size":7969177600,"raw":"\\\\.\\PHYSICALDRIVE1","displayName":"F:","protected":false,"system":false,"adapter":"standard","device":"\\\\.\\PHYSICALDRIVE1","mountpoints":[{"path":"F:"}],"description":"TS-RDF5 SD  Transcend USB Device"},"uuid":"21ae7a78-431e-4724-87e0-2357266677cd","unmountOnSuccess":true,"validateWriteOnSuccess":true})
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 sdk start +54s true
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 Sun Dec 10 2017 22:39:11 GMT+0000 (GMT Standard Time) Navigate ({"to":"success","from":"main"})
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 sdk stop +9s true
C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55 {"name":"Error","message":"Couldn't scan the drives: An unknown error occurred","cause":{},"isOperational":true,"stack":"Error: Couldn't scan the drives: An unknown error occurred"}
console.(anonymous function) @ C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\raven-js\dist\ra…:55
logException @ C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\resin-corvus\src…:228
exports.report @ C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\lib\gui\modules\exception-rep…:39
driveScanner.on @ app.js:218
emitOne @ events.js:96
emit @ events.js:191
onError @ C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\lib\shared\sdk\index.js:130
emitOne @ events.js:96
emit @ events.js:191
drivelist.listAsync.map.catch @ C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\lib\shared\sdk\standard\index…:79
tryCatcher @ C:\Users\Andrew\AppData\Local\Temp\nsm31D5.tmp\app\resources\app.asar\node_modules\bluebird\js\rele…:16
[... cut ...]
Steps to reproduce:
1. In the Etcher settings window, make sure "Eject on success" is enabled
2. Flash any image to a removable drive (in my case I was using a microSD card in a USB cardreader)
3. Wait for the flashing, valdating and unmounting to complete, and the success screen to display
4. Physically remove the USB drive
5. Etcher pops up the "Couldn't scan the drives" error dialog

P.S. I guess this is something else to add to https://github.com/resin-io/etcher/blob/master/docs/MANUAL-TESTING.md @jviotti 😉

lurch, 2017-12-10:
Oh! Seems like I don't even need to flash an image first; I can start Etcher, insert the drive, wait for Etcher to recognise the drive, then remove the drive, and it triggers the same error:
```
C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\raven-js\dist\raven.js:55 sdk scanner:init +0ms Scanner
C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\raven-js\dist\raven.js:55 sdk subscribe +0ms StandardAdapter
C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\raven-js\dist\raven.js:55 sdk subscribe +15ms USBBootAdapter
C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\raven-js\dist\raven.js:55  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@resin.io

Version = 1.2.1, Type = portable
C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\raven-js\dist\raven.js:55 Sun Dec 10 2017 23:10:49 GMT+0000 (GMT Standard Time) Application start ({"packageType":"portable","version":"1.2.1"})
C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\raven-js\dist\raven.js:55 Sun Dec 10 2017 23:10:49 GMT+0000 (GMT Standard Time) Checking for updates ({"currentVersion":"1.2.1","stable":true,"updateSemverRange":"<2.0.0","includeUnstableChannel":false})
C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\raven-js\dist\raven.js:55 sdk start +2s true
C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\raven-js\dist\raven.js:55 Sun Dec 10 2017 23:10:53 GMT+0000 (GMT Standard Time) Update notification skipped ({"reason":"Latest version"})
C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\raven-js\dist\raven.js:55 sdk stop +8s true
C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\raven-js\dist\raven.js:55 {"name":"Error","message":"Couldn't scan the drives: An unknown error occurred","cause":{},"isOperational":true,"stack":"Error: Couldn't scan the drives: An unknown error occurred"}
console.(anonymous function) @ C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\raven-js\dist\raven.js:55
C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\raven-js\dist\raven.js:55 Unhandled rejection TypeError: expecting an array or an iterable object but got [object Null]
    at apiRejection (C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\bluebird\js\release\promise.js:10:27)
    at MappingPromiseArray.init (C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\bluebird\js\release\promise_array.js:62:19)
    at Promise._settlePromise (C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\bluebird\js\release\promise.js:558:21)
    at Promise._settlePromise0 (C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\bluebird\js\release\promise.js:606:10)
    at Promise._settlePromises (C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\bluebird\js\release\promise.js:685:18)
    at Promise._fulfill (C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\bluebird\js\release\promise.js:630:18)
    at Promise._resolveCallback (C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\bluebird\js\release\promise.js:425:57)
    at Promise._settlePromiseFromHandler (C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\bluebird\js\release\promise.js:516:17)
    at Promise._settlePromise (C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\bluebird\js\release\promise.js:561:18)
    at Promise._settlePromise0 (C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\bluebird\js\release\promise.js:606:10)
    at Promise._settlePromises (C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\bluebird\js\release\promise.js:681:18)
    at Async._drainQueue (C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\bluebird\js\release\async.js:138:16)
    at Async._drainQueues (C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\bluebird\js\release\async.js:148:10)
    at Async.drainQueues (C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\bluebird\js\release\async.js:17:14)
    at <anonymous>
console.(anonymous function) @ C:\Users\Andrew\AppData\Local\Temp\nst98A2.tmp\app\resources\app.asar\node_modules\raven-js\dist\raven.js:55
```
This old Window7 laptop is a bit slow by modern standards (AMD E-350 Processor 1.60 GHz, 8.00 GB RAM) so perhaps something is blowing up inside `drivelist` when the drive gets removed at a point that drivelist doesn't expect the drive to be removed? Just a guess.

jhermsmeier, 2017-12-11:
@lurch can we move this to drivelist, as this is not strictly an issue with Etcher?

lurch, 2017-12-11:
As I said, it was only a _guess_ that it's a problem with `drivelist` itself, and not e.g. a problem with the way that Etcher uses drivelist. At the very least, it seems easy to reproduce this problem with Etcher - I wouldn't be sure how to reproduce this problem with drivelist standalone.

jhermsmeier, 2018-01-05:
Should be fixed in https://github.com/resin-io/etcher/pull/1953

closed 2018-01-05 by jhermsmeier