balena-io/etcher #1920: Etcher 1.2.1

Vladimir1302, 2017-12-13:
Etcher 1.2.1- **Etcher version:**
Linux ROSA Desktop Fresh KDE R10 X86_64- **Operating system and architecture:**
No finding file 
[image]
[image]

- **Image flashed:**
- **Do you see any meaningful error information in the DevTools?**

lurch, 2017-12-13:
I don't know if anyone on the Etcher team is able to read Russian - could you translate that error message to English for us please?

I did a quick search and apparently Linux ROSA uses rpm packages, so if you install Etcher using https://github.com/resin-io/etcher#redhat-rhel-and-fedora-based-package-repository-gnulinux-x86x64 then it should install all the required dependencies for you automatically?

I _think_ this issue might effectively be a duplicate of #835 ?

Vladimir1302, 2017-12-13:
I wrote the required program file to resolve dependencies not in OS LINUX ROSA DESKTOP FRESH KDE P10 but in the repository there is a similar file.Vladimir Sergeevich KalmetievSincerely,+79876261691vladimir.kalmetjev@yandex.ru

Vladimir1302, 2017-12-13:
I wrote the required program file to resolve dependencies not in OS Linux ROSA Desktop Fresh R10 but in the repository of the OS has a similar file --Sincerely,Vladimir Sergeevich Kalmetiev+79876261691vladimir.kalmetjev@yandex.ru

jhermsmeier, 2017-12-13:
The error dialog says:

> Some of the requested packages can not be imprinted (?): etcher-electron-1.2.1-1x86_64 (due to unsatisfactory libXScrnSaver) Still continue to install?

Vladimir1302, 2017-12-13:
The installation is not happening. --Sincerely,Vladimir Sergeevich Kalmetiev+79876261691vladimir.kalmetjev@yandex.ru

Vladimir1302, 2017-12-13:
In place of this file libXScrnSaver in the repository this file libxscrnsaver1 and libxscrnsaver-devel --Sincerely,Vladimir Sergeevich Kalmetiev+79876261691vladimir.kalmetjev@yandex.ru

jhermsmeier, 2017-12-13:
Oh I see, thank you! That's odd – @lurch any idea how that's having a "1" at the end of it in this case?

Vladimir1302, 2017-12-13:
This file is described OS a different name but all the letters in the name are the same as the requested program file to resolve dependencies only small and at the end added 1 --Sincerely,Vladimir Sergeevich Kalmetiev+79876261691vladimir.kalmetjev@yandex.ru

Vladimir1302, 2017-12-13:
Screenshot 1 request program 2 screnshot outdoor synaptic --Sincerely,Vladimir Sergeevich Kalmetiev+79876261691vladimir.kalmetjev@yandex.ru

lurch, 2017-12-13:
Hmmm, perhaps it's possible that in the Fedora repositories the package is named `libXScrnSaver` but in the [ROSA_Linux](https://en.wikipedia.org/wiki/ROSA_Linux) repositories the package is named `libxscrnsaver1` ?
I've no idea how "standardised" RPM package names are, or what we should do in this scenario :-/

Cheekily pinging @Dark-Passenger as he helped us with our RPM packaging before ;)

Vladimir1302, 2017-12-13:
I was wondering if it would be possible to allow to use the file from the repository of Rosa Linux, the program Etcher 1.2.1--Sincerely,Vladimir Sergeevich Kalmetiev+79876261691vladimir.kalmetjev@yandex.ru

Vladimir1302, 2017-12-13:
Rosa Linux is a close relative of Mandriva --Sincerely,Vladimir Sergeevich Kalmetiev+79876261691vladimir.kalmetjev@yandex.ru

Vladimir1302, 2017-12-13:
And Alt Linux BASEALT he is a close relative of OPEN SUSE--Sincerely,Vladimir Sergeevich Kalmetiev+79876261691vladimir.kalmetjev@yandex.ru

Vladimir1302, 2017-12-13:
In ALTLINUX BASEALT can come too and files from the FEDORA LINUX but most of it from OPEN SUSE . --Sincerely,Vladimir Sergeevich Kalmetiev+79876261691vladimir.kalmetjev@yandex.ru

Vladimir1302, 2017-12-13:
In Rosa Linux can run some files from the Fedora Linux but most of the files from Mandriva --Sincerely,Vladimir Sergeevich Kalmetiev+79876261691vladimir.kalmetjev@yandex.ru

Vladimir1302, 2017-12-13:
Here are the things that make Russian --Sincerely,Vladimir Sergeevich Kalmetiev+79876261691vladimir.kalmetjev@yandex.ru

Vladimir1302, 2017-12-13:
It is close to the universal OS --Sincerely,Vladimir Sergeevich Kalmetiev+79876261691vladimir.kalmetjev@yandex.ru

jhermsmeier, 2017-12-13:
By the way, @Vladimir1302 feel free to write in Russian – we will translate it somehow :)

zvin, 2017-12-13:
Добрый день Владимир, вы можете писать по Руский, я переведу.

Vladimir1302, 2017-12-13:
Роса Линукс сделан на основе Mandriva , Альт Линукс он же Basealt сделан на основе open SUSEно, поддержка файлов Федора в некоторой мере в них обоих есть надеюсь что это описание происхождения этих ОС поможет сделать возможнымработать программе Гравер в этих ОС так же как и ранее программе Яндекс Диск Индикатор поддерживаемой Github которая есть в их репозиториях этих ОС-- С уважением,Владимир Сергеевич Кальметьев+79876261691vladimir.kalmetjev@yandex.ru

zvin, 2017-12-14:
Translation of the above:

> Rosa Linux is based on Mandriva.
> ALT Linux (made by BaseALT) is based on Suse Linux.
> Both support Fedora packages to some extent.
> I hope this helps you making Etcher work on these OSes the same way yandex-disk-indicator [supported by / supporting] Github which exists in the repositories of these OSes.
> 
> With respect, Vladimir Sergeevitch

Vladimir1302, 2017-12-14:
Файл называется libXScrnSaver в:ALT Linux(BaseALT);Cent OS;Fedora;Slakvare .Файл называется libxscrnsaver1 в: Magea;open Mandriva;Rosa Linux .-- С уважением,Владимир Сергеевич Кальметьев+79876261691vladimir.kalmetjev@yandex.ru

zvin, 2017-12-14:
> The file is named libXScrnSaver in ALT LInux, CentOS, Fedora and Slackware.
> The file is named libxscrnsaver1 in Mageia, OpenMandriva and Rosa Linux.

lurch, 2017-12-14:
@zvin Cheers, could you please translate:

> Thank you Vladimir for your info and research. I'm now downloading `OpenMandrivaLx.3.03-PLASMA.x86_64.iso` so I'll have a fiddle around with that and see if I can come up with a solution.

zvin, 2017-12-14:
Ответ от @lurch :

> Большое спасибо @Vladimir1302 за ваше исследование и информацию.
> Я сейчас скачиваю `OpenMandrivaLx.3.03-PLASMA.x86_64.iso`, посмотрю смогу ли я найти решение.

lurch, 2017-12-14:
Hi Vladimir,

I've just been experimenting with OpenMandriva LiveCD in a VirtualBox, and it seems like if I download the `etcher-electron-1.2.1.x86_64.rpm` file I can install it using:
```
sudo rpm -i --nodeps etcher-electron-1.2.1.x86_64.rpm
```
and even though I don't have the `libxscrnsaver1` package installed, Etcher seems to run okay? (I tested inside VirtualBox by downloading an ISO file and flashing it to a virtual USB disk).
Does this solve the problem for you?

(Could do another translation please @zvin ?)

Vladimir1302, 2017-12-14:
Да ,это помогло установить программу и она работает. Я благодарю Вас всех за помощь.-- С уважением,Владимир Сергеевич Кальметьев+79876261691vladimir.kalmetjev@yandex.ru

Vladimir1302, 2017-12-14:
Yes,it helped to install the program and it works.I thank You all for the help. --Sincerely,Vladimir Sergeevich Kalmetiev+79876261691vladimir.kalmetjev@yandex.ru

lurch, 2017-12-15:
Hooray, glad we got there in the end @Vladimir1302 . Thanks for your translation help @zvin :-)
Closing this issue.

closed 2017-12-15 by lurch