balena-io/etcher #1922: Google Drive File Stream and user account elevation issue

m-p-3, 2017-12-14:
- **Etcher version:** 1.2.1
- **Operating system and architecture:** Windows 10 Pro 64-bit (1703)
- **Image flashed:** ubuntu-17.10-desktop-amd64.iso
- **Do you see any meaningful error information in the DevTools?**

**Description of the issue**
On a corporate system with Google Drive File Stream installed and connected to a G Suite account, it is currently impossible to flash a USB drive while the virtual network drive Google Drive File Steam (G:) is connected.

This issue occur mainly when the user runs the app without admin privileges and needs to elevate the app through UAC. Since the G: network drive is unreadable to other accounts, it shows this error:

```
Couldn't scan the drives: Permission eeor
Error: Couldn't scan the drives: Permission error
```

**Workaround**
Temporarily stop the Google Drive File Stream client if you need to elevate the app to write a USB drive, or completely sign out and login with an account with administrative rights on the system instead of proceeding with a *runas*.

Is that something that could be filtered out from the drive scan, or at least make USB drives to show up anyway?

lurch, 2017-12-14:
I assume the problem here is that the virtual drive created by "Google Drive File Stream" is only readable by the current user, and so when Etcher elevates to an admin-user, the `drivelist` part of Etcher then throws an error because it can't read this virtual drive.
Sounds similar to the problems we had with BitLocker etc. @jhermsmeier ?

lurch, 2018-02-28:
Does Etcher 1.3.1 (which uses a later version of the `drivelist` component) work any better in this situation?

jhermsmeier, 2018-02-28:
This is likely an issue with how the virtual storage of Google Drive File Stream is implemented (as a virtual file system rather than a block device). But we fixed that in drivelist for Windows a while back, so this should (theoretically) work fine with Etcher v1.3.1 now.
Would be great if you could let us know @m-p-3!

m-p-3, 2018-03-05:
Hi, I've tried it (back from paternity leave) and it works fine on my end. Thanks for your help!

jhermsmeier, 2018-03-05:
Sweet, thanks for the update @m-p-3! Closing this then

closed 2018-03-05 by jhermsmeier