balena-io/etcher #1923: Record Raspbian image on SDHC

SirFilbido, 2017-12-15:
- **Etcher version: 1.2.1**
- **Operating system and architecture: Windows 10 - x64**
- **Image flashed:**
[image]

Hello, I'm having this problem in creating the Respbian image. I'm using an 8GB SanDisk SDHC. I have already formatted the card and downed the image to see if something had been corrupted. Would you know how to solve this question?

lurch, 2017-12-15:
Is it possible that either your SD card or your card-reader is slightly faulty? Do you have a different card and/or a different card-reader you could try?

SirFilbido, 2017-12-15:
É possível que seu cartão SD ou o seu leitor de cartões esteja um pouco defeituoso? Você tem um cartão diferente e / ou um leitor de cartões diferente que você poderia tentar?

I did a scan in SD through H2testw and the result was this:

Warning: Only 7671 of 7672 MByte tested.
The media is likely to be defective.
3.8 GByte OK (8009408 sectors)
3.6 GByte DATA LOST (7700800 sectors)
Details:0 KByte overwritten (0 sectors)
0 KByte slightly changed (< 8 bit/sector, 0 sectors)
3.6 GByte corrupted (7700800 sectors)
0 KByte aliased memory (0 sectors)
First error at offset: 0x00000000f46d8000
Expected: 0x00000000f46d8000
Found: 0x0000000000000000
H2testw version 1.3
Writing speed: 6.12 MByte/s
Reading speed: 11.7 MByte/s
H2testw v1.4

Already the card reader, I believe that is not with problem since I am using of the laptop that is new. Unfortunately I can not immediately have another card to test or an external reader.

lurch, 2017-12-15:
`The media is likely to be defective.` and `3.6 GByte DATA LOST (7700800 sectors)` doesn't sound good.

Did you buy this card from eBay by any chance? From those H2testw results it sounds like you've bought a 4GB SD card which has been illegally reprogrammed to report its size as 8GB. If you can, try to get a refund.

SirFilbido, 2017-12-15:
But can this be affecting performance and giving this problem in the image recording?

jhermsmeier, 2017-12-15:
Yes. Those test results mean that half of the SD-card you have is broken and won't store any data.

jhermsmeier, 2017-12-15:
Closing this issue, as it's not an issue with Etcher, but with the storage device;
@SirFilbido feel free to keep the conversation going, I'm just closing this to avoid it getting in between bugs and such.

closed 2017-12-15 by jhermsmeier

lurch, 2017-12-15:
@SirFilbido You can read more about this at https://sosfakeflash.wordpress.com/