balena-io/etcher #1926: Etcher cannot differentiate folder and iso image when folder is named with .iso ending

lowks, 2017-12-16:
- **Etcher version: 1.2.1**
- **Operating system and architecture: Mac Mini 2012 running High Sierra **
- **Image flashed: openSuSE leap 42.3**
- **Do you see any meaningful error information in the DevTools? No**

When downloading OpenSuSE 42.3 image it will dump it in a folder called: `openSUSE-Leap-42.3-DVD-x86_64.iso`.  When using Etcher to pick the iso image it wrong assumes that the folder is the iso image and complains that the image is not valid.

lurch, 2017-12-21:
I'm not able to reproduce this on Linux - in fact the browse-file dialog doesn't allow me to select just a folder. I'll test on Windows later.
@jviotti Could you test this on your Mac to see if you can reproduce?

jhermsmeier, 2017-12-21:
This seems to be a bug in the Mac OS file open dialog (?), or more likely configuration thereof.

This is what it looks like in Finder (normal folder expansion):
[image]

While the open dialog doesn't allow expansion of the folder, due to the `.iso`:
[image]

jviotti, 2017-12-22:
I confirm I can reproduce on macOS. Ping @jhermsmeier. It should be a
quick fix.

jhermsmeier, 2018-01-19:
Fixed in #1976

closed 2018-01-19 by jhermsmeier