balena-io/etcher #1927: unable tp flash drive in Zorin 12

eltechoyqr, 2017-12-17:
- **Etcher version:**etcher-1.2.1-linux=x86_64
- **Operating system and architecture:**Linux workstation1 4.10.0-33-generic #37~16.04.1-Ubuntu SMP Fri Aug 11 14:07:24 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

- **Image flashed:**RetrOrangePi-4.1.Orangepione.img
- **Do you see any meaningful error information in the DevTools?**no info

etcher open recognizes the drive and the image file ... asks for password and then nothing 
dmesg | tail

 
[ 4180.990856] sd 4:0:0:2: [sde] 31285248 512-byte logical blocks: (16.0 GB/14.9 GiB)
[ 4180.997866]  sde: sde1 sde2 sde3
[ 5334.846655] sd 4:0:0:2: [sde] 31116288 512-byte logical blocks: (15.9 GB/14.8 GiB)
[ 5334.857040]  sde: sde1
[ 5335.288461] FAT-fs (sde1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
I have tried 3 different cards  ... tried fsck no change 
I can copy files to the disk in file manager 
gparted says 14.83G   fat 32

lurch, 2017-12-20:
I've just tried this myself using the Zorin 12 LiveCD running in VirtualBox to use the Etcher v1.2.1 AppImage to flash a virtual USB disk, and I didn't have any problems at all :-/

Could you please try this again, press `Ctrl` + `Alt` + `i` to display the DevTools, and copy and paste the entire Console output here please? That should at least allow us to see exactly **where** Etcher is getting stuck ;-)

I dunno if it'll help in your specific case, but sometimes Linux users report that Etcher works better for them if they run it with `sudo`.

eltechoyqr, 2017-12-22:
I tried again and this time the dev tools finally opened I did a screen 
capture and it is attached

I tried in a terminal window ( did not ask for my password )

  sudo etcher-1.2.1-linux=x86_64

and

sudo ./etcher-1.2.1-linux=x86_64

from inside /home/wayne/Downloads

both times it responded as follows

wayne@workstation1:~/Downloads$ sudo ./etcher-1.2.1-linux=x86_64
usage: sudo -h | -K | -k | -V
usage: sudo -v [-AknS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-AknS] [-g group] [-h host] [-p prompt] [-U user] [-u user]
             [command]
usage: sudo [-AbEHknPS] [-r role] [-t type] [-C num] [-g group] [-h 
host] [-p
             prompt] [-u user] [VAR=value] [-i|-s] [<command>]
usage: sudo -e [-AknS] [-r role] [-t type] [-C num] [-g group] [-h host] [-p
             prompt] [-u user] file ...

properties of file

-rwxrwxr-x 1 wayne wayne 79691776 Dec  6 14:10 etcher-1.2.1-linux=x86_64

I used an older version on a Zorin 9 system for a while and loved it

Etcher-1.0.0-rc.3-linux-x64.AppImage

Is the problem .AppImage is missing on the new one?

Sincerely

Wayne

lurch, 2017-12-22:
I'm curious as to how you've managed to get a file called `etcher-1.2.1-linux=x86_64` ?

Also, you can't send image attachments to GitHub via email (so I'm unable to see your DevTools screenshot), you need to login to the GitHub web-interface and attach them there.

eltechoyqr, 2017-12-22:
I originally downloaded from you site a while bake and simply double clicked on the downloaded file and it unzipped to the file as shown.
Today I again downloaded the same file and this time opened the archive manager and then located the file ...opened it...extracted it  and this is the result
wayne@workstation1:~/Downloads$ ll e*.*
-rw-r--r-- 1 wayne wayne     1193 Aug 31  2016 easy_OBD_test.txt
-rwxrwxr-x 1 wayne wayne 79691776 Dec  6 14:10 etcher-1.2.1-linux=x86_64*
-rw-rw-r-- 1 wayne wayne 76925843 Dec 22 12:21 etcher-1.2.1-linux-x86_64 (1).zip
-rw-rw-r-- 1 wayne wayne 76925843 Dec  8 19:22 etcher-1.2.1-linux-x86_64.zip
-rwxr-xr-x 1 wayne wayne 79691776 Dec  6 14:10 etcher-1.2.1-x86_64.AppImage*

as you can see the most recent download is (1) and they look similar 
I ran the .AppImage and flashed a sd card successfully
so I ran diff on the two extracted files with no difference showing up
I also attched the image of screen capture 
[image]
I used Pale Moon for the first download
Today I used Vivaldi (under test) for the download today

lurch, 2017-12-23:
Hmm, so perhaps this was due to some glitch of whatever "archive manager" you used the first time, which somehow messed the names up?
As everything seems to be working correctly now I'll close this issue, but feel free to leave any other useful info you can think of.

... actually no, I'll leave this open for now. Looking at your screenshot, I wonder if there's a bug in the way we're running the command which means that it'd actually fail if there's an `=` character anywhere in the file-path leading up to the name of the AppImage file? I'll do some testing after Christmas...

eltechoyqr, 2017-12-23:
last update ...
I renamed the original file with the = sign in the name ... changed it to the following
etcher-1.2.1-linux-x86_64
and then ran the file and it worked 100%
so the = sign in a file name is obviously the problem, not a limitation in linux I am aware of
unless it is the new kernel
Linux workstation1 4.10.0-33-generic #37~16.04.1-Ubuntu SMP Fri Aug 11 14:07:24 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

Foxtrod89, 2017-12-23:
My live USB flash drive's been locked after burning image with Etcher. After couple tricks with EaseUS i made it work, but Windows 10 recognize it as hard drive. Does anybody know how to fix that?

lurch, 2017-12-24:
@Foxtrod89 If you're having a problem, please create a [new issue](https://github.com/resin-io/etcher/issues/new) rather than 'hijacking' an existing issue.
But to answer your question, please see https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives

lurch, 2018-02-28:
@eltechoyqr I've just done some experimenting (see below), and the issue was indeed with just the `=` in the filename, which was confusing `sudo`. So closing as this isn't an Etcher problem.

```bash
$ echo -e '#!/bin/bash\necho Hello' > script_x86
$ chmod +x script_x86 
$ ./script_x86 
Hello
$ sudo ./script_x86 
Hello
$ mv script_x86 script=x86 
$ ./script=x86 
Hello
$ sudo ./script=x86 
usage: sudo -h | -K | -k | -V
usage: sudo -v [-AknS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-AknS] [-g group] [-h host] [-p prompt] [-U user] [-u user]
            [command]
usage: sudo [-AbEHknPS] [-r role] [-t type] [-C num] [-g group] [-h host] [-p
            prompt] [-u user] [VAR=value] [-i|-s] [<command>]
usage: sudo -e [-AknS] [-r role] [-t type] [-C num] [-g group] [-h host] [-p
            prompt] [-u user] file ...
```

closed 2018-02-28 by lurch