balena-io/etcher #1928: Unable to flash .iso to Sand Disk USB

cRenture, 2017-12-18:
Hello,

I downloaded Etcher 1.2.1 on my Kali Linux system and im trying to flash the latest version of Kali Linux 2017.3 .ISO to my Sand Disk USB Device but the flashing stops between 53 ~ 63 % and locks my USB Device to Read-Only Protection which i can't undo from the USB Option Menu in Kali Linux, so i had to download a tool called "Gparted" in Kali Linux and from there forcefully format it and create a fully new partion on the USB because it gives me error "unallocated" when im plugin in the USB in any computer. 

Is there anything i can do on my end or is there any other tool i can use on Kali Linux to flash the ISO to my USB?

cRenture, 2017-12-18:
Never mind! I fixed it. 

I downloaded Etcher from here on Github instead which had lot's of neccesary files for Etcher to work propertly. I also had to add the "Etcher debian repository" and "Trusted Bintray.com's GPG key" and install "etcher-electron". My bad. Etcher is an amazing tool! =))

lurch, 2017-12-18:
Ah cool, glad you got it sorted! :-)

closed 2017-12-18 by lurch